earlephilhower/arduino-pico #2671: Add support for CheckPicoW() for the Pico2W

erichelgeson, 2024-12-05:
Currently the [CheckPicoW()](https://github.com/earlephilhower/arduino-pico/blob/master/cores/rp2040/cyw43_wrappers.cpp#L116C13-L116C23) is guarded with `#ifdef PICO_RP2040` - else `true`. This will incorrectly return `true` on a Pico 2 (non-W).

It would be great if this runtime check could be made to work with each of the Pico/Pico-W/Pico2/Pico2-W versions.

I appreciate all the time and effort you put into maintaining this project, Thanks!

earlephilhower, 2024-12-05:
I commented out the PicoW check on the RP2350 boards because it uses some special analog tests that were valid on the PicoW.  I am *not* certain they work properly on the Pico 2W (don't have one yet) or if it works on boards with the CYW43 RP-WiFi module like the SparkFun Thing Pro RP2350.

If the RPi folks could provide some boilerplate code that would be really awesome here.  The original code came from one of their appnotes...

lyusupov, 2024-12-05:
Schematic:
1)  W https://datasheets.raspberrypi.com/picow/pico-w-datasheet.pdf
2) 2W https://datasheets.raspberrypi.com/picow/pico-2-w-schematic.pdf

GPIO 25 and 29 circuits looks identical

[image]

erichelgeson, 2024-12-05:
>  I am not certain they work properly on the Pico 2W (don't have one yet) 

I just got my Pico2W's yesterday. I pulled the `CheckPicoW()` code into my sample app and ran it and it fell into the `gp25` "cant tell" branch - so something seems to be different here unfortunately, even though the circuits look identical.

lyusupov, 2024-12-05:
> Edit: A bit more testing shows the ADC check seems to work fine, but the gp25 does not.

Q1 is FET. 
GPIO 25 is connected to it's Gate

I am curious why this code is trying to READ the state of this Gate rather than to set it to HIGH or LOW to switch the FET On or Off.

[image]

earlephilhower, 2024-12-06:
Check https://github.com/earlephilhower/arduino-pico/blob/0061d3f97fd4c610f735bd19caf74eb414da3d6d/cores/rp2040/cyw43_wrappers.cpp#L114-L115 for the source of the code.  I didn't dig into the specifics, sorry.

Lannakin, 2026-02-14:
[Soon...](https://github.com/Lannakin/rpi-pico-pizero-wireless-uhat-rpi-radio-module-2)....maybe.  I need to look over it some more, cuz I don't know what I'm doing.

Lannakin, 2026-02-21:
update: the prototype boards and resistors and stuff will be here in a week or two.  i can hardly wait to discover a horrible mistake in the design, at which point (if i can't do cute prototype board fixes with jumper wires) i will bolt the boards together and create a lovely $20 paperweight!

and, of course, if i turn out not to have gotten a paperweight, i guess i can do tests.

Lannakin, 2026-03-28:
update:  I swapped out the resister finally;  I had been putting it off until I got it leaded solder so it would be easier to rework any oopsies.

I used [this branch with it added as a variant](https://github.com/Lannakin/arduino-pico/tree/rm2-uhat-test) and [the attached code](https://github.com/user-attachments/files/26326174/Blink.txt) (it's not really saved as `.txt`, I promise).  I seem not to be getting a clock signal, though?

I guess I need to set up the STM32-based cheapo oscilloscope and look closer at whether or not I'm getting a clock and just not seeing any voltage changes on my multimeter, or what.

Lannakin, 2026-04-05:
well, i ended up using the 50 year old 15MHz oscilloscope from work to have a look at the "spi clk" signal.  I set up the board variant to use the default gpio pinout for the cyw43 interconnect.  waveshare connects gpio29 to the USB otg connector at ...um, whatever data line is the white wire.  it's got a few ohms of resistance on it because it's part of that usb differential pair or w/e, but I think I could remove the chip resistor that's also in the circuit and solder to the pad going to the chip if I had to because usb otg is uncool anyway.

anyway, looking at this signal in channel X with gnd as channel Y, this is how the waveform looks.

so I think I killed my radio module.  I've ordered a pimoroni rm2 breakout board to test this code on.*

*the real motivation is to see if I want to revise the newer revisions of my pcb before I send it in

[image]

(I was super excited that when I plugged it in, the chassis did not shock me, and the scope didn't let any magic smoke out.  idk if it's fast enough to actually be useful, and it definitely has some noise in the signals, presumably because the electrolytic caps have retired, but maybe it's still a little useful.  it also smells like all the old electronics at work tend to)