earlephilhower/arduino-pico #2674: Setting the RTC on start-up?

mcarrickscott, 2024-12-06:
#include <time.h>  
...
  Serial.println(__DATE__ );
  Serial.println(__TIME__ );  
  time_t rawtime;
  struct tm * timeinfo;

  time (&rawtime );
  timeinfo = localtime ( &rawtime );
  Serial.print ( "The current date/time is: ");
  Serial.println( asctime (timeinfo) );

outputs 

Dec  6 2024
10:16:34
The current date/time is: Thu Jan  1 00:03:06 1970

That last time increments if placed in a loop. So there is a real-time clock, but it does not initialize to the current time. Now with some awkward programming (Arduino does not support strptime() and the tm structure does not have a tm_gmtoff member) it would be possible to do this myself. But it seems that something that should happen by default ??

mcarrickscott, 2024-12-06:
Should have added that this arises with Raspberry Pi Pico W

earlephilhower, 2024-12-06:
The sketch needs to set the time on startup if it cares.  See the NTP examples and the BearSSL ones (since certificate verification needs the present UTC).

There's no battery backup for the RTC peripheral and it's not even used by the core to keep time anyway (only use I've seen is waking up from sleep on the ro2040) not does it exist in the rp2350.

closed 2024-12-06 by earlephilhower