earlephilhower/arduino-pico #2676: Auto Reboot to BOOTSEL mode - PICOTOOL

hulek-ales, 2024-12-11:
Hello, 
due security restrictions on my PC a can't use default upload method and I must use Picotool. 
But on my Arduino IDE 2.3.4 and older (WIN 11 PC) RP2040 didn't reboot to BOOTSEL mode automaticly and a have to press button and replug USB. 

So I added "tools.picotool.upload.use_1200bps_touch=true" into platform.txt and now RP enter to BOOTSEL automaticly.
Please check my solution. 
Bye

Gavin-Perry, 2025-02-06:
This issue is so close to what I want that I didn't open a new one. I'm looking for a simple utility that I can give to users of my Pico powered devices to download a UF2 file.

I read [Loading your RPI PICO without bootsel button](https://forums.raspberrypi.com/viewtopic.php?t=350232#p2099535)
and have downloaded OpenOCD but I'm stuck at that point. If some kind person could package a utility that will let a user connect to a Pico that is on a USB com port and download a selected file that would be great. Or just get the drive to open without pressing BootSel and the user can drag and drop. I know it's possible since the Arduino IDE does it but I don't want naïve users to have to deal with the whole IDE environment just to copy my UF2 file onto their device. Not to mention it would be handy in a production line to drop code into a set of devices.

Thanks for the support!

earlephilhower, 2025-02-06:
It's possible to program the flash using OpenOCD and the SWD connector, but frankly I think that's going to be way harder to get right than just drag-and-dropping a UF2 onto a "USB stick" (which the Pico will appear as, under practically all OSes).  The incantation used by the IDE is here
https://github.com/earlephilhower/arduino-pico/blob/5e2fbf324b8f55fb47c53e13b88b67caad445ab0/platform.txt#L220

Note this does NOT use the programmed Pico's USB *at all*.  You need a working DebugProbe hooked up to the SWD port on the Pico to be programmed, and to install the OpenOCD app/dependencies to get it running.

Why can't you just ship the UF2 file and ask the user to hold BOOTSEL and drag-and-drop the 1 file?  No drivers, no apps, no problem...

Gavin-Perry, 2025-02-06:
And by the way the problem with pressing the boot select button is that it's not available since the device is packaged in a box with other electronics.

Sent via the Samsung Galaxy S22 5G, an AT&T 5G smartphone

https://github.com/earlephilhower/arduino-pico/blob/5e2fbf324b8f55fb47c53e13b88b67caad445ab0/platform.txt#L220

earlephilhower, 2025-03-19:
Just tested w/latest release and can't reproduce.  Be sure to build and upload with picotool one time before this will work (the picotool USB interface isn't added unless you compile using the picotool upload option).