earlephilhower/arduino-pico #2688: Serial.print ignore for the first 3 seconds

naoppy, 2024-12-15:
- Environment
  - Raspberry Pi Pico2
  - connected with USB to PC
 
- Issue
  - Serial.print does not print anything for the first 3 seconds.

- The minimum code
```arduino
int cnt = 0;
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(9600);
  Serial.println(cnt++);
}
void loop() {
  Serial.println(cnt++);
  digitalWrite(LED_BUILTIN, HIGH);
  delay(1000);
  Serial.println(cnt++);
  digitalWrite(LED_BUILTIN, LOW);
  delay(1000);
}
```
and my output is
```
3
4
5
...
```

I’m a novice microcontroller user, so I do not know if these behaviors are unusual or common.

dquadros, 2024-12-15:
This is complete normal. It takes some time to establish the serial connection over USB. You can wait for the connection before doing a print:
 
```
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  while (!Serial) {
    delay(10);
  }
  Serial.begin(9600);
  Serial.println(cnt++);
}
```

naoppy, 2024-12-15:
Thanks for the quick reply, and thanks for sharing `while(!Serial)` syntax!

closed 2024-12-15 by naoppy

dhalbert, 2024-12-15:
Note that a common issue for sketches that might need to run when unconnected to USB is to wait forever on `while (!Serial)`. If you want the program to run without being connected to Serial, put some kind of timeout on waiting for `Serial`.

ldursw, 2024-12-19:
Here’s a snippet I use in my projects to handle USB connections. If your project doesn’t rely on USB all the time, you can skip the waiting process when it’s not connected.

```cpp
#include <device/usbd.h>
#include <Arduino.h>

void setup() {
    // Wait for USB to connect
    delay(500);

    // Check if USB is connected
    if (tud_connected()) {
        // Wait up to 5 seconds for serial to connect
        uint64_t start = millis();
        while (!Serial && millis() - start < 5000);
    }

    // If Serial is available, give the terminal a moment to initialize
    if (Serial) {
        delay(100);
    }

    Serial.println("Hello World");
    // ...
}

void loop() {
    // ...
}
```