earlephilhower/arduino-pico #2689: Can't compile blink example in case user home folder contains non ascii chars - e.g. german umlauts

Jueff, 2024-12-15:
Sketch: blink.ino
Arduino IDE: 2.3.4
Board version: 4.3.1

Build fails with:
C:/Users/**J�rgen**/AppData/Local/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file C:\Users\**Jürgen**\AppData\Local\arduino\sketches\D0046F654E9AC3258C0A12A3A49085A8/memmap_default.ld: Invalid argument
collect2.exe: error: ld returned 1 exit status
exit status 1
Compilation error: exit status 1

seams like some char conversion fails.

Jueff, 2024-12-15:
@earlephilhower: with Arduino Mbed OS RP2040 boards 4.2.1 blink.ino compiles correctly.

earlephilhower, 2024-12-16:
The issue isn't the LD, it's the command line being passed into it or a missing file that's not generated because the IDE's passing in something probably expected to be ASCII somewhere.  MBED doesn't do stuff like that since they don't support the filesystem or CPU options we do.

I can take a look into it but I don't have a DE language Windows so if you could enable full compile logs, rebuild from scratch in a new folder/run, and export the log to a TXT file you can attach here I can see if I can see anything that's obvious.

Jueff, 2024-12-16:
[buildlog.txt](https://github.com/user-attachments/files/18145796/buildlog.txt)
complete build log of blink.ino with comments

seams that g++ compiles and creates the files needed for linker, they are all present in the filesystem, but linker fails to read the input files correctly.

earlephilhower, 2024-12-16:
Thanks, good debugging.

I had hoped that the `simplesub.py` was not writing the file because of missing quotation marks, but on inspection all the command lines seem to be quoted properly and the file was actually created on your system.

What you discovered by going from the LFN to the 8.3 is something most likely in the MINGW porting layer or the GNU LD 14.1 linker Windows-compatibilty layer.  There's nothing we can do about that here, rewriting either of those is a little out of scope for this project. :laughing: 

Why the Arduino toolchain can link?  That's probably due to them using a different(older) MingW porting layer and a much older GNU toolset (7.x vs the 14.x we're on).

So, unfortunately, I think you're stuck with either using a username w/o non-ASCII chars or sticking with the Arduino MBED core.

earlephilhower, 2024-12-16:
One other option, of course, would be to see if Platform.IO performs any differently.  I believe there's even a VSCode integration so you get the same editor, more or less, as the Arduino IDE.

Jueff, 2024-12-16:
Thank you for the far-reaching analysis. We will not invent a new LD 😆
As a workaround I start arduino-builder for command line, specifying a 8.3 platform path.

`-prefs=runtime.platform.path=C:\Users\JRGEN~1\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.3.1`

I didn't find a way to set it somewhere in the Arduino IDE preferences.
But for now the workaround is ok.

earlephilhower, 2026-02-17:
Seeing as how Arduino was a European company before being bought, it surprises me the IDE isn't handling non-ASCII well.  Especially since it's the new `arduino-cli` that's doing all the work now, not any legacy or VSCode bits...

@febles it looks like you wrote a good update.  Why not do a PR to the docs with it so you get `git blame` credit. 😄