earlephilhower/arduino-pico #2694: Wifi-Channel-Number and Wifi-Power not settable/changeable on WiFi.mode(WIFI_AP)

Rob58329, 2024-12-16:
Using WiFi.mode(WIFI_AP) on the Pico2W:
(1) do not appear to be able to set the AP-wifi-channel-number (it always seems to use channel=3)
(2) the "WiFi.channel()" command always gives "channel=-1" [although it does work correctly for WiFi.mode(WIFI_STA)]  
(3) the "WiFi.aggressiveLowPowerMode()" and "WiFi.defaultLowPowerMode()" don't appear to do anything for WiFi.mode(WIFI_AP)

The code I'm using is: 
 ```
  WiFi.mode(WIFI_AP); delay(100);
  Serial.printf("softAP-Config=%s\n", WiFi.softAPConfig(hotspot_ip,hotspot_ip,(IPAddress){255,255,255,0}) ? "OK" : "FAIL"); delay(100);
  Serial.printf("WiFi.softAP=%s\n",WiFi.softAP(hotspot_ssid, hotspot_pw, 7)  ? "OK" : "FAIL"); delay(100); // setting channel (eg. to 7) does not work -> always uses channel=3 
  Serial.printf("channel=%i\n",WiFi.channel()); delay(100); // always gives "channel=-1"
  WiFi.aggressiveLowPowerMode(); delay(100); // does not appear to work
  // WiFi.defaultLowPowerMode(); delay(100); // does not appear to work
 
```
PS. I'm using Arduino-IDE v1.8.19 with "CPU Architecture=Arm" and "https://github.com/earlephilhower/arduino-pico" as at 16Dec24.

earlephilhower, 2024-12-16:
Good catch.  I think when I added WiFi back when it first came out (went to Microcenter to pick up the last available one on launch day!) the CYW43 driver did not have the option to set the AP channel.  It's been rev'd many times since and there does now seem to be a call `cyw43_wifi_ap_set_channel` which could be used here.

If you look in the code, it's thrown away in the ESP8266 wrapper but should be passed down and used in the `softAP` build.
https://github.com/earlephilhower/arduino-pico/blob/dc0dc50e36ee1b0ca69ab7017e141537cb9d2fc5/libraries/WiFi/src/WiFiClass.cpp#L162-L165

https://github.com/earlephilhower/arduino-pico/blob/dc0dc50e36ee1b0ca69ab7017e141537cb9d2fc5/libraries/WiFi/src/WiFiClass.cpp#L177

Re: the power modes, this is a limitation of the driver.  We don't distinguish between AP/STA modes and just call the CYW43's low level function no matter what.  The AP mode may simply not allow these kinds of power savings.

 https://github.com/earlephilhower/arduino-pico/blob/dc0dc50e36ee1b0ca69ab7017e141537cb9d2fc5/libraries/WiFi/src/WiFiClass.cpp#L523-L529

earlephilhower, 2024-12-16:
Oh, and `WiFi.channel()` is only for use when you are scanning for nearby APs.  See the `ScanNetworks` example.  It's just the way Arduino defined things (instead of doing another object for returning a found network) that we're trying to stay compatible with. `-1` == error == correct in this case, since you're not actually going through scan results.

earlephilhower, 2024-12-17:
Can you give #2695 a try?  A quick test with the CaptivePortal example seems to set the channel appropriately here.

Rob58329, 2024-12-17:
@earlephilhower
>Can you give #2695 a try?

This update work a treat, and setting the softAP wifi-channel on Pico2W now works fine - thank you!  :-)

PS. From "https://github.com/georgerobotics/cyw43-driver/blob/main/src/cyw43_ll.c" it looks like both of the power-saving modes just tinker with the beacon-intervals and similar to reduce average power consumption a bit (which I guess are unlikely to impact softAP-mode at all).

And there does not appear to be any controls for the max-output-power such as the ESP8266/ESP32
`"WiFi.setOutputPower(); / WiFi.setTxPower(WIFI_POWER_8_5dBm);"` commands - a pity!

commit 84694960 (2024-12-17): Support setting channel for SoftAP

Fixes #2694
commit 2d5f2da4 (2024-12-17): Support setting channel for SoftAP

Fixes #2694
PR #2695 (2024-12-17): Support setting channel for SoftAP
Fixes #2694
commit 33b12683 (2024-12-17): Support setting channel for SoftAP (#2695)

Fixes #2694
closed 2024-12-17 by Rob58329