earlephilhower/arduino-pico #2710: 8th servo channel rejected

dpharris, 2024-12-21:
Problem: Asking for the 8th servo channel is rejected with 'no PIO channel available'.  

Is this a 'one-off' error, or is some other system already using one of the PIO channels?   

Thanks

earlephilhower, 2024-12-21:
If you are using FreeRTOS, one PIO SM is used because the FreeRTOS task switching takes over some HW resources we need.

If you're not running FreeRTOS, can you please provide a simple MCVE that shows the problem?

earlephilhower, 2024-12-21:
````
#include <Servo.h>
Servo myservo[8];
void setup() {
  delay(5000);
  Serial.printf("Begin\n");
  for (int i = 0; i < 8; i++) {
    Serial.printf("%d = %d\n", i, myservo[i].attach(i));
  }
  Servo onetoomany;
  Serial.printf("onetoomany = %d\n", onetoomany.attach(8));
}
void loop() {
}
````

outputs the expected 8 servos with the 9th failing...
````
10:34:42.474 -> Begin
10:34:42.474 -> 0 = 0
10:34:42.474 -> 1 = 1
10:34:42.474 -> 2 = 2
10:34:42.474 -> 3 = 3
10:34:42.474 -> 4 = 4
10:34:42.474 -> 5 = 5
10:34:42.474 -> 6 = 6
10:34:42.474 -> 7 = 7
10:34:42.474 -> onetoomany = -1
````

closed 2024-12-27 by earlephilhower

dpharris, 2024-12-29:
Earle --
Thanks for the response, and for you libraries - much appreciated. 
It appears that Wifi uses one of the PIOs on the Pico, so that may explain why I was seeing a problem. 
Another issue maybe that I am declaring a servo array.  
Anyway, does not seem to be a problem with your library.  
Thanks again.  
David