earlephilhower/arduino-pico #2713: Board Support for DEFCONBadge32 (RP2350A)

MeteorTheProto, 2024-12-22:
## THE DC32
This board was created as a collab by a bunch of people including RPi themselves for DEFCON 32. This board is included in the PicoSDK [here](https://github.com/raspberrypi/pico-sdk/blob/master/src/boards/include/boards/defcon32_badge.h). This board has no RAM so I want to make sure that the compiler is aware of that. Attached are various resources for the board (schematic, images, ect). I have tried to make my own support for this board but it was much harder than I anticipated.

### Resources:

- [PicoSDK board defines file link](https://github.com/raspberrypi/pico-sdk/blob/master/src/boards/include/boards/defcon32_badge.h])
- [dc32_assembly_drawing.pdf](https://github.com/user-attachments/files/18220422/dc32_assembly_drawing.pdf)
- [dc32_badge_schematic.pdf](https://github.com/user-attachments/files/18220423/dc32_badge_schematic.pdf)
- [dc32_badge_test_procedure.pdf](https://github.com/user-attachments/files/18220424/dc32_badge_test_procedure.pdf)
- [dc32_block_diagram](https://github.com/user-attachments/assets/664a1097-90f2-403c-9af3-3a1507f70962)
- [dc32_bom.pdf](https://github.com/user-attachments/files/18220425/dc32_bom.pdf)
- [dc32-badge product page](https://shop.defcon.org/products/dc32-badge)

[image]

earlephilhower, 2024-12-22:
While it looks a neat bit of kit, I don't think there's a real need for adding a board definition file here.  AIUI, they were a one-off for DEFCON32, but it's not like anyone is producing them now.

Your best bet is to use the generic RP2350 with no PSRAM selected and 4MB flash.  With that you can use the core's IO with appropriate `SPI.setMOSI()/etc. calls to align with the schematic and multitude of peripherals.

closed 2024-12-22 by earlephilhower

MeteorTheProto, 2024-12-22:
Alright.

MeteorTheProto, 2024-12-22:
actually, is there any way to make the LEDS turn green on successful bootloader mode like adafruits's boards?

earlephilhower, 2024-12-22:
Das Blinkenlight seem to be WS2812s, so if that's what you're talking about you need to use a 3rd party WS control library and run it at the beginning of `setup`.  If you're talking about a plain LED somewhere else (didn't see it on the schematic but it's early here...)  then identify the GPIO pin and do the usual `pinMode(Xxx, OUTPUT); digitalWrite(xxx, LOW/HIGH);` at `setup` start.