earlephilhower/arduino-pico #2718: Bluetooth audio

qinzian2000, 2024-12-25:
When using Pico 2W to run sketch A2DPSink.ino, I found that Pico 2W has a lot of noise and can only play one lyric, causing Bluetooth to disconnect. Is there any way to solve this problem

earlephilhower, 2024-12-25:
Thanks, I was able to verify this happens on the RPI WiFi module on the SparkFun Thing Plus RP2350 (no Pico 2Ws at Microcenter yet) and verify also the the exact same code works fine on the Pico W (I actually left my phone connected to the PicoW overnight and it was still playing).

Given that IIRC there is no code difference in the code in *this* repo for A2DP for the Pico vs. Pico 2, my worry is that this is an issue in the Pico SDK.  The BT stack has always been fragile, in my experience, and maybe there's just something in the Pico2 driver that's not playing nice with the higher data rates of A2DP.

Have you experienced any issues with standard BT or BLE?  Or just the A2DP?

I think the next step is to build the A2DP sample using the raw SDK and see if that also craps out on the Pico2W.  If so, it'll be out of our hands but we can open an issue on the SDK repo.

earlephilhower, 2025-01-13:
Are you saying the A2DPSink here works or the BTStack example does?  And I assume you ran for a few minutes.  I think I got something like 3-4 mins of uptime and then things kind of fell apart when trying on the SparkFun RP2350 board with the code here (but I don't remember if it was PWM or I2S).

pschatzmann, 2025-01-13:
I was testing your A2DPSink example on a Pico 2W: I just changed the output from PWM to I2S!
Initially I just played one song (for a couple of minutes). 

I therefore repeated the test and let it run for a bit more then 15 minutes w/o having any issues. 
I also moved the phone to the next room w/o loosing any connection.

Some additional info: was testing with Android! (On the ESP32 iOS does also strange things with the combination of A2DP and I2S)

So the issue might be maybe PWM or maybe iOS or a bad Pico 2W ?

earlephilhower, 2025-01-13:
Great, thanks for the info!

I only have what may be preproduction 3rd party boards with RPI-WiFi daughtercards so there's some slight chance they weren't quite up to snuff.  I'll give it one more go (I was doing Android streaming as well) when I get a chance and try I2S as well as PWM.

qinzian2000, 2025-01-14:
Thank you all for your help

earlephilhower, 2025-01-14:
I can confirm that the Pico 2W I2S + A2DPsink runs fine, while the PWMAudio + A2DPSink results in a crash on the Pico2W.  There's also a `#ifdef` issue in the latest master (not in a release yet) but that's orthogonal to this.  Even removing the load from the PWMPins (i.e. assuming it's an electrical thing) doesn't change the results.

earlephilhower, 2025-03-19:
I've just tried this with the current head and 2.1.2-develop on the SDK.  With no code changes I've run `A2DPSink.ino` on the SparkFun ThingPlus RP2350 with PWM output for about 10 minutes now.  Looks like this was a SDK/BTStack issue that was fixed with the latest release...no related code here has changed as far as I can see.  So, closing this a fixed...

closed 2025-03-19 by earlephilhower