earlephilhower/arduino-pico #2724: fs::File as class member crash

hellcats88, 2024-12-29:
Hi,

I have a strange behavior using the fs::File class in my Pico project.
I have this class definition:
```
#include <SD.h>
class db {
  public:
    db();
    db(const db &o) = delete;

    bool load_new_books(err_t &err);

  private:
    fs::File idxPf;
};
```

In the source file this is the function:
```
#include "db.h"
#include <SD.h>

bool db::load_new_books(err_t &err) {
  this->idxPf = SD.open(DB_INDEX_FILE, "a+");
  if(!this->idxPf.isFile()) {
    sprintf(err.m, "Failed to open %s index on filesystem", DB_INDEX_FILE);
    return false;
  }
  ...
}
```

When I upload the code in my Pico it freeze during the ``SD.open`` function.
I have found 2 workarounds: 
1. move the fs::File idxPf variable from the class definition to the source file as local variable:
```
bool db::load_new_books(err_t &err) {
  fs::File idxPf = SD.open(DB_INDEX_FILE, "a+");
  if(!idxPf.isFile()) {
    sprintf(err.m, "Failed to open %s index on filesystem", DB_INDEX_FILE);
    return false;
  }
  ...
}
```
2. move the fs::File idxPf variable from the class definition to the source file as module variable
```
fs::File idxPf;

bool db::load_new_books(err_t &err) {
  idxPf = SD.open(DB_INDEX_FILE, "a+");
  if(!idxPf.isFile()) {
    sprintf(err.m, "Failed to open %s index on filesystem", DB_INDEX_FILE);
    return false;
  }
  ...
}
```

Can you help me to find the error? Honestly I prefer to used a class member variable...

earlephilhower, 2024-12-29:
Sorry, we need a real complete MCVE that shows the issue to look into it.  In general the `File` class (I believe there's a `using` statement so no need to call out `fs::File` explicitly) uses smart pointers to keep everything safe and sane so you can pass it around without issue.

A small app with a do-nothing object showing the failure would be the best way forward.  If you can't make that, then you can hook up a DebugProbe and get a stack dump on the crash to see where things are and if there's been memory corruption, etc.

hellcats88, 2024-12-29:
I have tried with a simple code as you say and the problem not occurs, so I will try to investigate in my project to check the problem.
When/if I find the reason I will update the issue for trace it. 

Thanks for your attention.

earlephilhower, 2024-12-30:
Closing for now.  Just make a note if you can repro with an MCVE and we can reopen.

closed 2024-12-30 by earlephilhower

hellcats88, 2024-12-31:
I've found the problem.
In the main file i was creating the object using
```
std::shared_ptr<db> storage();
```

instead of
```
std::shared_ptr<db> storage(new db());
```