earlephilhower/arduino-pico #2727: CPU halt when using some pins with Adafruit_NeoPixel

YeWuChen, 2024-12-31:
I'm having some issues using the Adafruit_NeoPixel library.
My LED lights up fine when I run the strandtest using GPIO25, but when I use GPIO44 the RP2350 seems to freeze and Windows will pop up a notification saying USB Device not recognized.
Below is my test environment:
Generio RP2350B with 4M of Flash and 8M of PSRAM@109MHz using GPIO0 as CS.
CPU: ARM, 150MHz
Arduino-Pico Version: 4.4.0
Adafruit_NeoPixel library version: 1.11.0

earlephilhower, 2024-12-31:
This would be better handled in the NeoPixel repo.  I would imagine that the library needs to be updated to work properly with the RP2350B and the add'l GPIO pins above 32 which have special needs w/PIO and direct GPIO  access.

closed 2024-12-31 by earlephilhower

YeWuChen, 2024-12-31:
Ok, thank you very much!