earlephilhower/arduino-pico #2729: Error compilation with last release

pierrotm777, 2025-01-01:
Hello,

I just compile with the last release and have error:

`In file included from c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\BackgroundAudio\src/BackgroundAudio.h:26,
                 from C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\SoundModules_Pierre\RP2040\Sound_Christian_RP2040_Ph-Motor\Sound_Christian_RP2040_Ph-Motor.ino:218:
c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\BackgroundAudio\src/BackgroundAudioMixer.h: In member function 'uint32_t* BackgroundAudioMixerInput::getResampledBuffer()':
c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\BackgroundAudio\src/BackgroundAudioMixer.h:248:34: error: 'LED_BUILTIN' was not declared in this scope
  248 |                     digitalWrite(LED_BUILTIN, LOW);
      |                                  ^~~~~~~~~~~
c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\BackgroundAudio\src/BackgroundAudioMixer.h:264:34: error: 'LED_BUILTIN' was not declared in this scope
  264 |                     digitalWrite(LED_BUILTIN, LOW);
      |                                  ^~~~~~~~~~~`

No issue with the 1.0.0 version

earlephilhower, 2025-01-01:
That's not this core, that's the [BackgroundAudio](https://github.com/earlephilhower/BackgroundAudio) library.  Your board doesn't have an LED pin defined (it's standard on all Arduino branded boards and most 3rd party ones). so you can safely remove those lines.  They were actually just debugging indicators I used on my own Pico and will be gone next rev of that library.

closed 2025-01-01 by earlephilhower

pierrotm777, 2025-01-01:
Thanks for your answer. Ok