earlephilhower/arduino-pico #2740: Problem with TFT_eSPI and board_build.core = earlephilhower

sartor, 2025-01-06:
Hi. I have problem with ST7789 display with `board_build.core = earlephilhower` setting on in platformio.ino
If I comment this line and use default core everything is working fine (but not working other stuff in my code)
But if board_build.core set to earlephilhower code compiles but display is blank.
Also I see only 2 warnings while compilation going. While on default core there are tons of warning in TFT_eSPI.c file

My platformio.ini:
```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
monitor_speed = 115200
lib_deps =
  bodmer/TFT_eSPI @ ^2.5.43

build_flags =
  -Os
  -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG
  -DUSER_SETUP_LOADED=1
  
  ; Define the TFT driver, pins etc. here:
  -DST7789_DRIVER=1
  -DTFT_SPI_PORT=0
  -DTFT_WIDTH=240
  -DTFT_HEIGHT=240
  -DTFT_MOSI=3
  -DTFT_SCLK=2
  -DTFT_DC=8
  -DTFT_RST=4
  -DLOAD_GLCD=1
  -DLOAD_FONT2=1
  -DLOAD_FONT4=1
  -DLOAD_FONT6=1
  -DLOAD_FONT7=1
  -DLOAD_FONT8=1
  -DLOAD_GFXFF=1
  -DSMOOTH_FONT=1
  -DSPI_FREQUENCY=27000000
```
My main.cpp:
```
#include <Arduino.h>

#define BLACK 0x0000
#define WHITE 0xFFFF
#define GREY  0x5AEB

#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();

void setup(void) {
  tft.init();
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK);
  tft.setTextColor(TFT_WHITE, GREY);
  tft.setTextSize(2);
  tft.setCursor(10, 10);
  tft.println("Hello!");
}

void loop() {
}
```

The only thing I change between working and not working display is `board_build.core = earlephilhower` config line

TobiasVanDyk, 2025-01-07:
Try
lib_deps =
  bodmer/TFT_eSPI @ ^2.5.34

sartor, 2025-01-07:
Hi. Thank you for suggestion. I've tried `bodmer/TFT_eSPI @ ^2.5.34` and `bodmer/TFT_eSPI @ 2.5.34` - same result, not working, blank screen

sartor, 2025-01-07:
```
 *  Executing task: C:\Users\user\.platformio\penv\Scripts\platformio.exe run --target upload --target monitor --environment pico 

Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; board: pico; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Library Manager: Installing bodmer/TFT_eSPI @ 2.5.34
Unpacking  [####################################]  100%
Library Manager: TFT_eSPI@2.5.34 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.15.0+sha.39b9039) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, pico-debug, picoprobe, raspberrypi-swd)
PACKAGES:
 - framework-arduinopico @ 1.40300.0+sha.e16c459
 - tool-mklittlefs-rp2040-earlephilhower @ 5.100300.230216 (10.3.0)
 - tool-openocd-rp2040-earlephilhower @ 5.140200.240929 (14.2.0)
 - tool-picotool-rp2040-earlephilhower @ 5.140200.240929 (14.2.0)
 - toolchain-rp2040-earlephilhower @ 5.140200.240929 (14.2.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
PSRAM size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 68 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TFT_eSPI @ 2.5.34
|-- SPI @ 1.0
Building in release mode
Compiling .pio\build\pico\FrameworkArduinoBootloader\boot2_w25q080_2_padded_checksum.S.o
Compiling .pio\build\pico\src\main.cpp.o
Generating linkerscript D:\MEGA\Coding\fpv-monitor\.pio\build\pico/memmap_default.ld
Compiling .pio\build\pico\lib1ee\SPI\SPI.cpp.o
Compiling .pio\build\pico\lib915\LittleFS\LittleFS.cpp.o
Compiling .pio\build\pico\lib915\LittleFS\lfs.c.o
Compiling .pio\build\pico\lib915\LittleFS\lfs_util.c.o
Compiling .pio\build\pico\libaf1\TFT_eSPI\TFT_eSPI.cpp.o
Compiling .pio\build\pico\FrameworkArduino\BluetoothDebug.cpp.o
Compiling .pio\build\pico\FrameworkArduino\Bootsel.cpp.o
Compiling .pio\build\pico\FrameworkArduino\CoreMutex.cpp.o
Compiling .pio\build\pico\FrameworkArduino\FS.cpp.o
Compiling .pio\build\pico\FrameworkArduino\PIOProgram.cpp.o
Compiling .pio\build\pico\FrameworkArduino\RP2040Support.cpp.o
Compiling .pio\build\pico\FrameworkArduino\RP2040USB.cpp.o
Compiling .pio\build\pico\FrameworkArduino\SerialPIO.cpp.o
In file included from src\main.cpp:7:
.pio\libdeps\pico\TFT_eSPI/TFT_eSPI.h:973:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
  973 |       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
      |        ^~~~~~~
In file included from .pio\libdeps\pico\TFT_eSPI\TFT_eSPI.cpp:16:
.pio\libdeps\pico\TFT_eSPI\TFT_eSPI.h:973:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
  973 |       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
      |        ^~~~~~~
Compiling .pio\build\pico\FrameworkArduino\SerialUART.cpp.o
Compiling .pio\build\pico\FrameworkArduino\SerialUSB.cpp.o
Compiling .pio\build\pico\FrameworkArduino\StackThunk.cpp.o
Compiling .pio\build\pico\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\pico\FrameworkArduino\WMath.cpp.o
Archiving .pio\build\pico\lib1ee\libSPI.a
Compiling .pio\build\pico\FrameworkArduino\_freertos.cpp.o
Compiling .pio\build\pico\FrameworkArduino\api\Common.cpp.o
Compiling .pio\build\pico\FrameworkArduino\api\IPAddress.cpp.o
Compiling .pio\build\pico\FrameworkArduino\api\PluggableUSB.cpp.o
Compiling .pio\build\pico\FrameworkArduino\api\Print.cpp.o
Compiling .pio\build\pico\FrameworkArduino\api\Stream.cpp.o
Compiling .pio\build\pico\FrameworkArduino\api\String.cpp.o
Compiling .pio\build\pico\FrameworkArduino\cyw43_wrappers.cpp.o
Compiling .pio\build\pico\FrameworkArduino\delay.cpp.o
[... 96 lines cut ...]
Verifying Flash:    [=================             ]  59%
Verifying Flash:    [==================            ]  62%
Verifying Flash:    [===================           ]  65%
Verifying Flash:    [====================          ]  68%
Verifying Flash:    [=====================         ]  71%
Verifying Flash:    [======================        ]  75%
Verifying Flash:    [=======================       ]  78%
Verifying Flash:    [========================      ]  81%
Verifying Flash:    [=========================     ]  84%
Verifying Flash:    [==========================    ]  87%
Verifying Flash:    [===========================   ]  90%
Verifying Flash:    [===========================   ]  93%
Verifying Flash:    [============================= ]  97%
Verifying Flash:    [==============================]  100%

  OK

The device was rebooted to start the application.
--- Terminal on COM12 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
Starting...
Init completed!
```

sartor, 2025-01-07:
Hi. Thank you for your suggestion. Suddenly `2.5.34` (not `^2.5.34`) works fine with `board_build.core = earlephilhower`!
Is there some regression bug in TFT_eSPI library?

earlephilhower, 2025-01-07:
This seems like it would be better handled in Bodmer's repo since it's all related to that library (or maybe just twitchy wiring?).  If there is really an issue in the core feel free to open another with an MCVE for it to be looked at.

closed 2025-01-07 by earlephilhower

TobiasVanDyk, 2025-01-08:
Is there some regression bug in TFT_eSPI library?
Yes see:
https://github.com/Bodmer/TFT_eSPI/discussions/3561
https://github.com/Bodmer/TFT_eSPI/issues/3564
https://github.com/Bodmer/TFT_eSPI/issues/3402
https://github.com/Bodmer/TFT_eSPI/issues/3476

sartor, 2025-01-08:
Oh! Thank you very much!