earlephilhower/arduino-pico #2741: RP2350 equivalent of __isPicoW?

jhmaloney, 2025-01-07:
The RP2040 had a global variable __isPicoW set to true at power up (in cores/rp2040/cyw43_wrappers.cpp) if the board has a cyw43 chip. It is returned by isPicoW() defined in RP2040Support.h. However, that variable is not defined when I compile the code for an RP2350 Pico2-W board.

That feature is very useful for MicroBlocks, an educational programming language. It allows us to distribute a single firmware binary that works on both Pico and Pico-W boards. That avoids potential confusion since users cannot accidentally install the wrong firmware for their board. Many of our users are new to microcontrollers so the difference between Pico and Pico-W boards is not as obvious to them as it is to more experienced users.

Is there an equivalent way to detect the presence of the cyw43 on the RP2350? If not, what happens if you try to use the cyw43 on a board that does not have one?

lyusupov, 2025-01-08:
Duplicates https://github.com/earlephilhower/arduino-pico/issues/2671

jhmaloney, 2025-01-08:
Glad you and Earle are already working on this. You can close this issue; I will follow [#2671](https://github.com/earlephilhower/arduino-pico/issues/2671).

closed 2025-01-08 by earlephilhower