earlephilhower/arduino-pico #2742: __uninitialized_ram doesn't appear to work on pico2

sanjay900, 2025-01-08:
I suspect the fixes from https://github.com/earlephilhower/arduino-pico/issues/1186 need to also be applied to the 2350

earlephilhower, 2025-01-08:
Unfortunately not possible.  The 2350 randomizes SRAM on startup as part of the secure boot process, overwriting all old data from prior reset.

closed 2025-01-08 by earlephilhower

sanjay900, 2025-01-08:
Just out of curiosity is that documented somewhere?

sanjay900, 2025-01-08:
Because i applied the same changes as #1186 to the memory map for the 2350 and it seemed to work, so I just want to know why it seems to work when it shouldn't. I don't plan on using any of the secure boot features, so if this is only an issue when those are enabled then it would still be nice to have that fixed.

sanjay900, 2025-01-08:
Does this still apply if i am doing a reset via the watchdog and not a full reset? I was using `__uninitialized_ram` to store a variable and then resetting the pico via the watchdog, so it would still be good to have this fixed for that scenario since it does still seem to work in that case.