earlephilhower/arduino-pico #2743: I2S Multiple Inputs in 2024

thestumbler, 2025-01-08:
I have code working, based on various examples, using one I2S instance (connected to a PWM1802 ADC). Am trying to expand it to a two, or potentially four, input system. My first attempt isn't going well, and before I dig too deep into things, I'd like to confirm that it is even possible. For example, the issues raised in #535 are no longer a concern nowadays?

earlephilhower, 2025-01-08:
As far as we know, they work fine.  If you have a simple MCVE showing otherwise, then we can take a look.

thestumbler, 2025-01-08:
Thanks. I will make one this morning. Points I am suspicious of:

(1) interrupts, callbacks, and timings. 

(2) using an RP2350 vs RP2040

(3) GPIO pin selection and I2S channel initialization

thestumbler, 2025-01-09:
Okay, things are looking good. I had accidentally forgotten to start the second I2S input engine when refactoring the code. This issue is closed.

closed 2025-01-09 by thestumbler