earlephilhower/arduino-pico #2746: Multiple inheritance error when attempting to use A2DPSource as a Stream

tahosa, 2025-01-09:
I'm attempting to use the the A2DPSource as a destination for a stream copy using the utility from [arduino-audio-tools](https://github.com/pschatzmann/arduino-audio-tools/blob/main/src/AudioTools/CoreAudio/StreamCopy.h). Maybe I'm misunderstanding since it's been a long time since I've worked with C++, but `StreamCopy` should accept anything that derives from `arduino::Print` as a target. `A2DPSource` does appear to extend that interface, but when I try to use it, I get a compiler error that there is an ambiguous base:

```
error: 'arduino::Print' is an ambiguous base of 'A2DPSource'
```

I have added `arduino-audio-tools` v1.0.1 as a zip library in the Arduino IDE along with adding `arduino-pico` using the board manager.

Here is a simple sketch that reproduces the error:

```
#include <AudioTools.h>
#include <BluetoothAudio.h>

I2SStream in;
A2DPSource out;
StreamCopy copier(out, in);

void setup() {
}

void loop() {
}
```

Is there something I'm missing, or need to explicitly cast? I have tried to do so, but `arduino::Print` and `arduino::Stream` are both abstract classes and I'm not sure what concrete class I would need to cast to in order for this to work.

earlephilhower, 2025-01-09:
Sorry, but I really don't know anything about that library.  You may want to ping the author there and see what his suggestion is.

It's possible there's some weird C++ multiple inheritance issue here, but none of the examples in this repo are throwing errors and this is the first report like this.

tahosa, 2025-01-11:
Thanks for replying. I'll follow up with them if I need to, but I think I've figured out a way to efficiently copy straight from the I2S to the BT module. Now I just have to try to get I2S to correctly and completely lock onto the clock signal.

closed 2025-01-11 by tahosa