earlephilhower/arduino-pico #2750: I2S Clock Drift

tahosa, 2025-01-12:
In testing trying to use I2S to Bluetooth, I've been running into issues with just getting noise on the line so took a step back and have been investigating the raw I2S send/receive behavior. 

Using two PicoWs (one flashed to transmit, one to receive) I flash the two sketches below and connect them up using a breadboard and cheap jumpers matching the data, bclk, and lrclk pins (0 to 0, 1 to 1, 2 to 2) and watch the serial output of the receiver.

What I see is that instead of just getting a constant (1024, 0) for the L/R I2S data, it slowly drifts one bit at a time (1024, 0) -> (512, 0) -> (256, 0) -> (128, 0) -> ... -> (0, -32768) -> ... -> (0, 1) -> (-32768, 0) -> ... etc. etc.

Am I missing something from the documentation or examples that would keep the clock signals in sync, or is this a function of cheap breadboard/wires?

### Sketches

Transmit:
```
#include <I2S.h>

I2S i2s(OUTPUT);

void setup() {
  Serial.begin(115200);

  i2s.setDATA(0);
  i2s.setBCLK(1); // Note: LRCLK = BCLK + 1
  i2s.setBitsPerSample(16);
  i2s.setFrequency(44100);
  i2s.setSysClk(44100);
  i2s.begin();

  while (1) {
    i2s.write16(1024, 0);
  }
}

void loop() {}
```

Receive:
```
#include <I2S.h>

I2S i2s(INPUT);

void setup() {
  Serial.begin(115200);

  i2s.setDATA(0);
  i2s.setBCLK(1); // Note: LRCLK = BCLK + 1
  i2s.setBitsPerSample(16);
  i2s.setFrequency(44100);
  i2s.setSysClk(44100);
  i2s.begin();

  int16_t l, r, lastL, lastR;
  uint32_t count = 0;

  while (1) {
    lastL = l;
    lastR = r;
    i2s.read16(&l, &r);
    if(l != lastL || r != lastR) {
      Serial.printf("change at cycle\t%d\t%d\t%d\n", count, l, r);
      count = 0;
    }
    count++;
  }
}

void loop() {}
```

earlephilhower, 2025-01-12:
What you're doing isn't legal, you can't connect the I2S output from one Pico to the I2S input of another.  Both PicoWs must be masters and always reference their own internal clocks for sampling.  There will be some difference in each Pico's internal PLL (you're guaranteed 133,000,000 +/- X%, not exactly 133,000,000 cycles per second). So even if you're exceedingly lucky and get things lined up at the beginning, there's no guarantee they'll stay that way.

For clock jitter, that's because for most I2S frequencies there's no even divider from the PLL to drive the PIO so a fractional one is used which will add or remove a sysclk period as needed to keep the average correct.  There's a call to ensure the Pico runs at an integer divisor of the samplerate, see https://github.com/earlephilhower/arduino-pico/blob/master/docs/i2s.rst#bool-setsysclkint-samplerate

As far as I2S stability, in my own recent testing of BackgroundAudio I had the web radio running on the PicoW over I2S for over 12 hours without issue.  It's very simple and deterministic.

If you're getting clicks or pops there could be underflows (not generating data fast enough or being stuck in an IRQ too long and missing a transmit) or noise on the pins or I2S analog portion.

tahosa, 2025-01-12:
Thank you so much for that detailed explanation! That really helps me understand the underlying issue. I had already tried using `setSysClk()` and did get slightly different average de-sync intervals, but the detail about the relative difference between the two Picos really clarifies what's happening. I'll trust that my basic implementation for the I2s -> Bluetooth is good enough to move forward testing the actual I2S source I'm interested in and going from there.

You and this repo has been a huge help! Thanks for keeping it maintained and for all the work you do educating the community!

closed 2025-01-12 by tahosa