earlephilhower/arduino-pico #2752: SDFS.h Line 287

Palingenesis, 2025-01-12:
I had to change:  
"`return _opened ? _fd->availableSpaceForWrite() : 0;`"  
to:  
"`return _opened ? _fd->availableForWrite() : 0;`"
  
I was not able to compile some code from "Freenove_Bipedal_Robot_Kit_for_Raspberry_Pi_Pico-main"
Any code that involve producing sound failed to compile.  
  
Using Arduino IDE 1.8.xx to upload to Pico w, via COM port.
Data uploaded with "Pico LittleFS Data Upload."

earlephilhower, 2025-01-12:
Can you please make an MCVE that shows the issue?  I can't reproduce any build failure and the call does exist in ESP8266SDFat
````
$ grep -r availableSpaceForWrite libraries/
libraries/SDFS/src/SDFS.h:        return _opened ? _fd->availableSpaceForWrite() : 0;
libraries/ESP8266SdFat/src/FatLib/FatFile.cpp:int FatFile::availableSpaceForWrite() {
libraries/ESP8266SdFat/src/FatLib/FatFile.h:  int availableSpaceForWrite();
````

Palingenesis, 2025-01-13:
The only Minimum Complete ? Example is:
A gift I got for Christmas.
Download the zip from [Freenove](https://freenove.com/tutorial) choose FNK0033  Freenove Bipedal Robot Kit for Raspberry Pi Pico (W).  
There is an Instruction manual, follow instructions.

Using Arduino IDE 1.8.xx and Raspberry Pi Pico W:
Unzip Sketches.
Install "Pico LittleFS Data Upload"
Install Libraries.
Chose any sketch that has music. (Compile Fails: Exit 1)
Do change as I say in first post.
It now compiles and uploads.

I hope that is plain enough. It is what I did.
I have not used the pico for any other projects, so I have no other experience with the pico.
I just did what I did to get it working.

If the call does exist then may be its missed in a library install for Arduino IDE 1.8.xx, I used library manager to instal the libraries, I did not unzip the files provided in the zip.

This is my list of URLs:
https://github.com/tenbaht/sduino/raw/master/package_sduino_stm8_index.json  
https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json  
https://espressif.github.io/arduino-esp32/package_esp32_index.json  
http://arduino.esp8266.com/stable/package_esp8266com_index.json  
https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json  
https://raw.githubusercontent.com/dbuezas/lgt8fx/master/package_lgt8fx_index.json  

Like I say, first time with Pico.  
Used the compile error message to get it working.
Thought I should mention it.

Oh I am on (F**k every open source program) Windows 11.

earlephilhower, 2025-01-13:
What is the exact error message you get?  I'm not really able to download and build an undefined 3rd party library My best guess is you're running an incompatible SDFat library and there will probably be warnings about that in the build process.  The incompatible, not included w/this library version doesn't have that call, but the ESP8266SDFat distributed with arduino-pico does.  If you build with the incompatible version, SDFS will not work.  If you have `SDFat` installed thru the library manager or manually or it's part of that 3rd party kit, you can't use it and must remove it.

earlephilhower, 2025-01-13:
I just googled the name and they're actually shopping my Esp8266audio as well.  That definitely works on the pico without changes (there is CI in that repo on the 8266, 32, and pico).  So, check for and remove any SdFAT libraries you have out of this core.

closed 2025-01-13 by earlephilhower

Palingenesis, 2025-01-13:
I suppose I should have just used the zipped libraries, instead of using Arduino IDE Library Manager to get the latest from the web?  
  
The issue that the Arduino IDE showed was:  
In file included from C:\Users\name\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.4.1\libraries\SD\src/SD.h:24, from E:\Users\name\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.h:25, from E:\Users\name\Documents\Arduino\libraries\ESP8266Audio\src\AudioFileSourceSD.cpp:21: C:\Users\name\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.4.1\libraries\SDFS\src/SDFS.h: In member function 'virtual int sdfs::SDFSFileImpl::availableForWrite()': C:\Users\name\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.4.1\libraries\SDFS\src/SDFS.h:287:31: error: 'using std::__shared_ptr_access<File32, __gnu_cxx::_S_single, false, false>::element_type = class File32' {aka 'class File32'} has no member named 'availableSpaceForWrite'; did you mean 'availableForWrite'? 287 | return _opened ? _fd->availableSpaceForWrite() : 0; | ^~~~~~~~~~~~~~~~~~~~~~ | availableForWrite "?

If you can't reproduce it then close the issue.
I just believe in reporting things, it's how things are found.
It is probably an issue on my end, I have such a lot of libraries installed for Arduino IDE, and not every one renames there versions of files used in there libraries, I often see multiple copies the compiler has to choose from.

earlephilhower, 2025-01-13:
This bit says that you've got a non-supported SDFat library in your libraries directory (via ZIP or library manager) somewhere:  ` class File32' {aka 'class File32'} has no member named 'availableSpaceForWrite'`  .  That member exists in the SDFat shipped here. https://github.com/earlephilhower/ESP8266SdFat/blob/9e1457101ca47b9ace198086dbd8854efc573ccf/src/FatLib/FatFile.h#L399

Palingenesis, 2025-01-14:
OK if it exists, It looks like I need to wipe my whole Arduino Library as I don't know who/what is the culprit, and reinstall the libraries.  
Thank you for looking into it.
I don't use the pico much, It was a Christmas gift. Most of my projects are ESP/STM projects.  
I did a quick fix, and I thought I should mention it.
Perhaps if I had just copied the zip files into the library folder the issue may not have arisen.
I thought I was doing the right thing by using the library and boards manager.
Next time I will just copy the files given.

If you feel there is no issue close this.  

I just posted it because I thought I had done every thing the way it should be done.
I even repeated the install (Library manager) and nothing fixed it.  

If I have something else installed that the compiler is using I have no clue to what it is, but.
I did a windows search for "availableSpaceForWrite" in the "Arduino/library" folder and "Users/..../Arduino15/packages" folder  
The results where:  
Reference found in two main places,  
"Users/..../Arduino15/packages/rp2040/hardware/rp2040/4.4.1/libraries" in several sub folders. (src folder has the FatFile .c and .h)  
and  
"Users/..../Arduino15/packages/esp8266/hardware/esp8266\3.1.2\libraries\" in several sub folders. (src folder has the FatFile  .c and .h)  

Since the member "int availableSpaceForWrite();" is in both board installations (FatFile), it looks like I need a clean install of Arduino, (Again Again Again ...... ).

So I am clueless to the issue I have.