earlephilhower/arduino-pico #2754: I2S.h and ADCInput.h frequencies are interacting

Geekachuqt, 2025-01-14:
It seems the I2S and ADCInput libraries are somehow interacting, with the ADCInput library being affected by the samplerate set on the I2S library. 

I've attached a sketch showing the issue below. In it, I've created an ADCInput object with the frequency set to 48000, and an I2S object with the frequency set to 96000. Using the while(adc.available()) method of sampling, I've added a counter that increases once per loop of the method, and added a boolean statement that outputs millis() on the serial bus when it reaches the sample rate. This means it should trigger once per second. However, it is triggering twice per second instead.

```
#include <ADCInput.h>
#include <I2S.h>

#define I2S_WORDS 64
#define I2S_DATA_PIN 9
#define I2S_BCLK_PIN 10
#define I2S_BUFFERS 4
#define SAMPLE_RATE 48000

I2S i2s_output(OUTPUT);

ADCInput adc(A0, A1, A2, A3);

int16_t A0Buffer[I2S_WORDS] = { 0 };
int16_t A1Buffer[I2S_WORDS] = { 0 };
int16_t A2Buffer[I2S_WORDS] = { 0 };
int16_t A3Buffer[I2S_WORDS] = { 0 };

uint16_t wPointer = 0;

float fCheck = 0;

void setup() {

  pinMode(23, OUTPUT);
  digitalWrite(23, HIGH);

  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);
  Serial.begin(115200);

  i2s_output.setFrequency(SAMPLE_RATE*2);
  i2s_output.setDATA(I2S_DATA_PIN);
  i2s_output.setBCLK(I2S_BCLK_PIN);
  i2s_output.setBitsPerSample(16);
  i2s_output.setBuffers(I2S_BUFFERS, I2S_WORDS);

  adc.setFrequency(SAMPLE_RATE);
  adc.setBuffers(I2S_BUFFERS, I2S_WORDS);

  i2s_output.begin();
  adc.begin();

}

void loop() {

  while(adc.available()){
    fCheck++;

    A0Buffer[wPointer] = adc.read();
    A1Buffer[wPointer] = adc.read();
    A2Buffer[wPointer] = adc.read();
    A3Buffer[wPointer] = adc.read();

    i2s_output.write(A0Buffer[wPointer]);
    i2s_output.write(A0Buffer[wPointer]);

    wPointer++;
    wPointer %= I2S_WORDS;

    if(fCheck == SAMPLE_RATE){
      Serial.println("One Second");
      Serial.println(millis());
      fCheck = 0;
    }
  }
}

```

earlephilhower, 2025-01-14:
Thanks for the MCVE, but it produces samples at the same rate with or without I2S.  I've commented out any I2S reference in your sketch
````
#include <ADCInput.h>
//#include <I2S.h>

#define I2S_WORDS 64
#define I2S_DATA_PIN 9
#define I2S_BCLK_PIN 10
#define I2S_BUFFERS 4
#define SAMPLE_RATE 48000

//I2S i2s_output(OUTPUT);

ADCInput adc(A0, A1, A2, A3);

int16_t A0Buffer[I2S_WORDS] = { 0 };
int16_t A1Buffer[I2S_WORDS] = { 0 };
int16_t A2Buffer[I2S_WORDS] = { 0 };
int16_t A3Buffer[I2S_WORDS] = { 0 };

uint16_t wPointer = 0;

float fCheck = 0;

void setup() {

  pinMode(23, OUTPUT);
  digitalWrite(23, HIGH);

  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);
  Serial.begin(115200);

  //i2s_output.setFrequency(SAMPLE_RATE*2);
  //i2s_output.setDATA(I2S_DATA_PIN);
  //i2s_output.setBCLK(I2S_BCLK_PIN);
  //i2s_output.setBitsPerSample(16);
  //i2s_output.setBuffers(I2S_BUFFERS, I2S_WORDS);

  adc.setFrequency(SAMPLE_RATE);
  adc.setBuffers(I2S_BUFFERS, I2S_WORDS);

  //i2s_output.begin();
  adc.begin();

}

void loop() {

  while(adc.available()){
    fCheck++;

    A0Buffer[wPointer] = adc.read();
    A1Buffer[wPointer] = adc.read();
    A2Buffer[wPointer] = adc.read();
    A3Buffer[wPointer] = adc.read();

    //i2s_output.write(A0Buffer[wPointer]);
    //i2s_output.write(A0Buffer[wPointer]);

    wPointer++;
    wPointer %= I2S_WORDS;

    if(fCheck == SAMPLE_RATE){
      Serial.println("One Second");
      Serial.println(millis());
      fCheck = 0;
    }
  }
}
````

And the output with timestamps is "One Second" every 500ms:
````
13:47:30.824 -> 14002
13:47:31.320 -> One Second
13:47:31.320 -> 14502
13:47:31.817 -> One Second
13:47:31.817 -> 15002
13:47:32.313 -> One Second
13:47:32.346 -> 15502
13:47:32.842 -> One Second
13:47:32.842 -> 16002
13:47:33.338 -> One Second
13:47:33.338 -> 16502
````

earlephilhower, 2025-01-14:
So this actually did show a problem in the ADC sampling rate when multiple inputs were active.  It's nothing to do with I2S, but does show on any ADCInput with more than 1 sampled input.  #2755 fixes this.

Simple test of ADCInput only shows the error and fix
````
#include <ADCInput.h>

#define I2S_WORDS 32
#define I2S_DATA_PIN 9
#define I2S_BCLK_PIN 10
#define I2S_BUFFERS 8
#define SAMPLE_RATE 1000

ADCInput adc(A0, A1, A2, A3);

void setup() {
  Serial.begin(115200);
  adc.setFrequency(SAMPLE_RATE);
  adc.setBuffers(I2S_BUFFERS, I2S_WORDS);
  adc.begin();
  delay(3000);
}

uint32_t samples = 0;
void loop() {
  while (adc.available()) {
    int x = adc.read();
    if (samples < 8) Serial.printf("%d ", x);
    samples++;
    if (samples == 4 * SAMPLE_RATE) {
      Serial.printf("%d\n", millis()/1000);
      samples = 0;
    }
  }
}
````

With fix, 4 x samplerate per second
````
15:48:45.145 -> 13 4095 1340 2009 19 4091 1336 2009 12
15:48:46.138 -> 29 4095 1347 2010 14 4095 1331 2011 13
15:48:47.131 -> 15 4095 1357 2010 20 4095 1338 2012 14
15:48:48.157 -> 16 4095 1345 2012 24 4095 1347 2009 15
15:48:49.149 -> 10 4095 1358 2010 20 4095 1342 2011 16
````

Without fix, 16 x samplerate per second (4x too fast)
````
15:49:37.970 -> 14 4095 1409 2012 25 4095 1416 2009 3
15:49:38.235 -> 25 4095 1421 2009 17 4095 1410 2011 4
15:49:38.466 -> 5 4095 1422 2008 17 4095 1414 2012 4
15:49:38.731 -> 12 4095 1422 2009 25 4095 1424 2011 4
15:49:38.996 -> 19 4095 1414 2013 26 4095 1419 2010 4
15:49:39.227 -> 19 4095 1416 2013 14 4095 1421 2010 5
15:49:39.492 -> 18 4095 1419 2010 27 4095 1410 2010 5
15:49:39.724 -> 15 4095 1420 2010 16 4095 1428 2010 5
15:49:39.989 -> 8 4082 1423 2011 26 4095 1426 2012 5
15:49:40.220 -> 12 4095 1414 2013 24 4091 1424 2010 6
15:49:40.485 -> 12 4095 1432 2010 14 4095 1425 2010 6
15:49:40.750 -> 11 4095 1426 2010 9 4095 1420 2014 6
15:49:40.981 -> 17 4095 1424 2012 20 4095 1428 2011 6
15:49:41.246 -> 25 4095 1426 2011 21 4095 1417 2011 7
15:49:41.478 -> 12 4095 1430 2010 24 4095 1421 2013 7
15:49:41.742 -> 24 4092 1433 2010 14 4095 1424 2014 7
15:49:41.974 -> 20 4094 1428 2009 20 4095 1421 2013 7
15:49:42.239 -> 15 4095 1420 2012 8 4095 1423 2010 8
15:49:42.471 -> 18 4095 1428 2010 14 4095 1420 2013 8
15:49:42.736 -> 18 4095 1428 2010 18 4095 1420 2013 8
15:49:42.967 -> 21 4095 1426 2012 12 4095 1429 2010 8
````

closed 2025-01-14 by earlephilhower

Geekachuqt, 2025-01-15:
> Also, that's probably way too small of a buffer for 48khz (doubly so for 96khz!). At 96khz, a 64 word buffer only takes 64/96000 = 0.67 milliseconds between interrupts. Total space before underflow would be less than 2.5ms (i.e. if you get a long interrupt at the wrong time, it's underflows everywhere and actually performing the IRQ takes time, too).

Ah, maybe this is why I'm seeing and hearing subtle phase shifting on the I2S output compared to the input. All in all it has been working quite fine though - in my actual application I'm dedicating core0 to doing only input and output, while core1 does all the processing. I will try increasing the buffersize to see if I notice any difference. Thanks for the investigation, and sorry for misattributing the root cause - I assumed it was some kind of crosstalk because it worked as expected if I set the I2S frequency to be equal to the ADC frequency, but I guess what was actually happening was that the I2S read calls were blocking the loop from executing at rates higher than the I2S frequency.

PR #2755 (2025-01-14): Fix ADCInput clocks for multiple inputs
When multiple inputs were active, the frequency was being scaled two times resulting in incorrect sampling speed.  Correct to only scale the calculation and not the stored value (which is used in `begin`).

Fixes #2754
commit 0655b7d5 (2025-01-14): Fix ADCInput clocks for multiple inputs (#2755)

When multiple inputs were active, the frequency was being scaled two
times resulting in incorrect sampling speed.  Correct to only scale
the calculation and not the stored value (which is used in `begin`).

Fixes #2754