earlephilhower/arduino-pico #2759: Universal Binary Support

sfe-SparkFro, 2025-01-16:
It's possible to create a "universal binary" such that a single firmware file can be uploaded to the RP2040 or RP2350 (ARM and RISC-V), and it works on each processor. An example for the Pico SDK is here: https://github.com/raspberrypi/pico-examples?tab=readme-ov-file#universal

Most of the magic is done in the CMake files. As I understand it, it compiles one binary for each platform, then links all the binaries together in a single file. Once the firmware is uploaded, the boot ROM searches for the right binary for that device's processor, then executes from there. Some additional `#ifdef`'s may be needed to ensure proper compilation on each platform, and can also be used to determine the platform at runtime. For example:

```
#ifdef RP2040
bool platform_is_rp2040 = true;
#else
bool platform_is_rp2040 = false;
#endif
```

Is there currently a way to do this in the Arduino framework? Perhaps a user could add a similar CMake file to the sketch folder or something? It would also be great if there was a button or something in the Tools dropdown menu to enable "universal binary compilation" so projects can be flashed to any RP2xxx board.

earlephilhower, 2025-01-16:
That's very interesting but not really achievable in the Arduino IDE.  `arduino-cli` controls the build process completely (including things like automatically adding include paths, and defining the output files, etc.).

From the command line it's possible to build for different FQBNs simply enough.  You could automate that using `arduino-cli` and then take the raw objects and perform the extreme unction the current examples do.  You probably could even do the arduino-cli calls from within the CMakefile.txt itself so a single `cmake ..; make -j` could generate the UF2.

sfe-SparkFro, 2025-01-16:
Good to know, thank you! Will have a play with that.

closed 2025-03-18 by earlephilhower