earlephilhower/arduino-pico #2772: [Q] Is the compile option -DSDFAT_FILE_TYPE=3 predefined?

guidol70, 2025-01-22:
As I do compile for the Pico RunCPM normal with 
SDFAT_FILE_TYPE 1
for using File32 from SdFat

I do get a predefined warning since the last RP2040-Cores (sure 4.42 and maybe 4.41):

```
Generating function prototypes...

in this and following command is included -DSDFAT_FILE_TYPE=3:

"C:\\Users\\guido\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\4.0.1-8ec9d6f/bin/arm-none-eabi-g++" -I "C:\\Users\\guido\\AppData\\Local\\Temp\\arduino_build_685133/core" -c -Werror=return-type -Wno-psabi -DUSBD_PID=0x000f -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico 2\"" -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -DFILE_COPY_CONSTRUCTOR_SELECT=FILE_COPY_CONSTRUCTOR_PUBLIC -DUSE_UTF8_LONG_NAMES=1 -DSDFAT_FILE_TYPE=3 -DDISABLE_FS_H_WARNING=1 "-DARDUINO_VARIANT=\"rpipico2\"" -DPICO_FLASH_SIZE_BYTES=4194304 

C:\Users\guido\Desktop\RunCPM_v6_7\RunCPM_v6_7_Pico2_300Mhz_Internal-64k_22012025\RunCPM_v6_7_Pico2_300Mhz_Internal-64k_22012025.ino:28:9: warning: "SDFAT_FILE_TYPE" redefined
   28 | #define SDFAT_FILE_TYPE 1           // Uncomment for Due, Teensy or RPi Pico
      |         ^~~~~~~~~~~~~~~
<command-line>: note: this is the location of the previous definition
Compiling libraries...
Compiling library "SPI"
```

Normally I had only to change for the Pico the SDFAT_FILE_TYPE in SdFatConfig.h
(other boards (non avr) from 3 to 1)

With my predefine to 1 before inlcuding SdFat.h my RunCPM worls, because with the setting 3
RunCPM wont compile and 3 doesnt support the File32 option :(

guidol70, 2025-01-22:
The SdFatConfig.h has the follwing part for the SDFAT_FILE_TYPE:

```
#ifndef SDFAT_FILE_TYPE
#define SDFAT_FILE_TYPE 1
#else  // defined(__AVR__) && FLASHEND < 0X8000
// All other boards.
#define SDFAT_FILE_TYPE 3
#endif  // defined(__AVR__) && FLASHEND < 0X8000
#endif  // SDFAT_FILE_TYPE
```
and for other (than avr) boards for RunCPM all Users did change the
`#define SDFAT_FILE_TYPE 3`
to 
`#define SDFAT_FILE_TYPE 1`

With my predefine ist does work, but I would know where this setting was from, because it did work in the past with original SdFat (up to v2.3.0) and the ESP8266SdFat witout this prefine.warning.

But I also changed my SDFAT-FILE-TYPE-part in the SdFatConfig.h to
```
#ifndef SDFAT_FILE_TYPE
#define SDFAT_FILE_TYPE 1
// #else  // defined(__AVR__) && FLASHEND < 0X8000
// All other boards.
// #define SDFAT_FILE_TYPE 3
// #endif  // defined(__AVR__) && FLASHEND < 0X8000
#endif  // SDFAT_FILE_TYPE

// #define SDFAT_FILE_TYPE 3 // for STM32 SBCs
```

earlephilhower, 2025-01-22:
Sorry, but I don't really understand the question.

Before we used ESP8266SdFat which I forked off from SdFat years ago to rewrite the ESP8266 SD filesystem.  It was manually (painfully) merged with upstream SdFat changes on an intermittent basis.  That old fork had some values added pre-set in that config file, but those settings were never present in upstream SdFat.  Now, we don't touch a single file in SdFat and use the `-D` compiler options to set the values that this core needs for `SDFS` and `SD` to work.  The files in the core now are pristine, fresh from the factory bits with the sticker and plastic still on. 😆 

If you want to edit your copy, feel free to do so.  But you'll probably break `SD` and `SDFS` in the process.  If you don't use them, it doesn't really matter then.

guidol70, 2025-01-22:
As I understand it I never used SD and SDFS in the past and moved a long time ago from the ESP8266SDFat to the original SDFat from Greiman.
This - when not using SD and SDFS of the RP2040 Core - worked without problems/warnings
(maybe only the FS_H warning).

Now you are using the -D compile options which seem to override the original SdFat  (and there the SDFatConfig.h) 
(but does work with a only predefine-error when settting the SDFAT_FILE-TYPE before including the SdFat.h)

I could life with this warning :)

It also does work for the RunCPM compile when usinf this line in the platform.txt:

`build.sdfatdefines=-DFILE_COPY_CONSTRUCTOR_SELECT=FILE_COPY_CONSTRUCTOR_PUBLIC -DUSE_UTF8_LONG_NAMES=1 -DSDFAT_FILE_TYPE=1 -DDISABLE_FS_H_WARNING=1`

earlephilhower, 2025-01-22:
Ah, I see the default value is actually what we want (this differs from the ESP8266SdFat fork) so I'll drop the `SDFAT_FILE_TYPE` define from the platform.txt command line.  Then your redefinition warning should go away, too.

PR #2773 (2025-01-22): Don't set SDFAT_FILE_TYPE, default is OK
Fixes #2772

No need to set SDFAT_FILE_TYPE=3 as that is the defaulr value with upstream SdFat.  Remove it from platform.txt and platform.io build.
commit 84826935 (2025-01-22): Don't set SDFAT_FILE_TYPE, default is OK (#2773)

* Don't set SDFAT_FILE_TYPE, default is OK

Fixes #2772

No need to set SDFAT_FILE_TYPE=3 as that is the defaulr value with upstream
SdFat.  Remove it from platform.txt and platform.io build.

* Codespell got all techy
closed 2025-01-22 by earlephilhower