earlephilhower/arduino-pico #2781: Assumptions on (fs::File).fullName() not being full

tobozo, 2025-01-29:
hi,

thanks for maintaining this 👍 

A library project I'm working on uses Arduino's `fs::FS` and `fs::File` as common grounds for RP2040, ESP32 and ESP8266, it implements a recursive directory parser to build a list of files and folders under a given path. That parser uses `openNextFile()` in a loop, and goes recursive when the returned item is a folder.

Cross device portability is done with a couple of C macros that will use either `fs::File::path()` (ESP32) or `fs::File::fullName()` (RP2040/ESP8266).

Assumptions (mostly based on experiences with esp32): 

When `rootdir = fs.open("/")`, any `fs::File` returned by `rootdir.openNextFile()` should have its `::fullName()` prefixed by `/`.
e.g. if `fs::File blah` exist and is in the root folder, then `blah.fullName()` should be `/blah.txt`, not `blah.txt`.

I can't store `filename = file.fullName()` and use it later with `fs.open(filename)` as it will produce a `file not found` error.

So I'm wondering if this is some legacy behaviour I should address by prefixing the `/` myself, or if this is a bug in LittleFS where `/` should be prefixed to `::fullName()` when the directory is `/`.

earlephilhower, 2025-01-29:
LittleFS should not care if you try and `LittleFS.open("/path/to/file.txt", "r");` or ``LittleFS.open("path/to/file.txt", "r");`.  The initial `/` should always be implied with file names since there's not really any `chdir` in Arduino-land (nor does LittleFS even have a `lfs_chdir` that I can find).

I wouldn't mind a PR that prefixes a `/` in the `fullName` if it's not present, but I'm not seeing where it would make any difference.  The `fullname` should  be valid either way...

Could you give a small MCVE that shows the problem?  That would help me track down what you're seeing and why...

tobozo, 2025-01-29:
Thanks for your reply 👍 

The context of my project is to compress LittleFS contents to a .tar.gz archive. That archive can eventually be decompressed on a chdir capable system, where the difference between relative and absolute paths may matter.

It tried to prefix the `/` at runtime to every collected path, but still got the errors.

I figured the `file not found` error I was getting was caused by a lock in LittleFS e.g. when opening a file from core0 while another file was waiting to be closed on core1.

So I was just misled by my habits with esp32 core. I thought the error was caused by the lack of leading slash, but it's not.

It's still a good reminder to not expect perfect consistency across arduino cores :) 

Should I close this now that my assumptions are fixed?

earlephilhower, 2025-01-29:
Thanks for the update.  I'll close this one then.

FWIW, accessing the filesystem in parallel from both cores is not going to work very well without much care.

Like most libraries, the LittleFS lib doesn't explicitly support multicore parallel access.  For example, one core could update the filesystem structures at the same time another is updating them, leading to corruption.  If you protect all calls yourself (with a separate mutex so that only 1 core at a time calls anything which could update state), then it should work.  But if you miss a mutex around a call it could cause corruption to the filesystem and a crash or data loss at some point later on.

closed 2025-01-29 by earlephilhower

tobozo, 2025-01-31:
> not going to work very well without much care

Thanks for your hints 👍 I solved that by using FreeRTOS queues. The goal was to handle `tar.gz` compression without intermediate tar file in [ESP32-targz](https://github.com/tobozo/ESP32-targz/tree/LZ77?tab=readme-ov-file#pack--compress-to-targz-stream-no-intermediate-tar-file), I'll be producing a release soon.

May I request some feedback from you? I'm not 100% sure there's no side effects when including `FreeRTOS.h` from deep within the library e.g. I couldn't get `SingleFileDrive.h` from the USB-MSD example to work properly when `ESP32-targz.hpp` was included before.

The only function in ESP32-targz to use FreeRTOS tasks is `TarGzPacker::compress()`, and I don't want the library to cripple existing RP2040 projects just because of that.

earlephilhower, 2025-01-31:
We don't use FreeRTOS at all in the core by default.  It's bare metal SDK which keeps things fast and lean (but also less convenient than having tasks and queues).

When anyone includes FreeRTOS, the guts of the core start a completely different code path for things like USB, interrupts, timers, Newlib locking primitives, networking, `setup/loop`, etc.  So there are *lots* of side effects if FreeRTOS is included anywhere.  Most standard things work fine, but special USB interfaces like SingleFileDrive uses may not have ever been tested or function, unfortunately.

earlephilhower, 2025-01-31:
You can't really do tasks or queues without an OS.  You would start something in `setup1` (on the 2nd core) which could use either the `rp2040.fifo.push/pop` to get a work pointer from core 0 and do stuff.  You could also use global variables or the usual 1-writer 1-reader lockless fifo (see the UART implementations).

tobozo, 2025-02-01:
Thanks for your input, I'll keep that in mind for my pico projects.

I dropped the tasks/queues and used piped streams instad, now `SingleFileDrive` works fine so I guess it's settled.

It wasn't a good idea to add FreeRTOS as a dependency to a multidevice library anyway, lesson learned :)