earlephilhower/arduino-pico #2801: Incorrect flash for XIAO RP2350

ianm8, 2025-02-15:
Hi Earle

I'm developing a project around the Seeed Xiao RP2350 and noticed that the flash size has been incorrectly set at 16MB whereas it should be 2MB. (It is a 16 megabit device). It seems the project builds and programs the device ok.

Also, the IP stack is available and set to IPv4. Not sure if that is an issue since the device does not support networking?

Cheers, Ian

earlephilhower, 2025-02-15:
That's a simple fix, thanks for noting it.  I also see it is listed as having 8MB PSRAM in the boards.txt, but don't see any message about that on the web page (and can't find a schematic for it either).  Do you know if the PSRAM setting should also be removed?

ianm8, 2025-02-15:
No PSAM, that is also a mistake.

Cheers, Ian

PR #2803 (2025-02-16): Correct Seeed XAIO RP2350 config
Fixes #2801
commit 3cb5c315 (2025-02-16): Correct Seeed XAIO RP2350 config (#2803)

Fixes #2801
closed 2025-02-16 by earlephilhower