earlephilhower/arduino-pico #2802: Optional redefine of initVariant() outside of arduino-pico for PicoW&Pico2W

Mictronics, 2025-02-15:
`initVariant()` is initialized weak an can be redefined outside arduino-pico. However for the PicoW and Pico2W the `initVariant` is redefined and called to initialize the CYW43 driver via call `init_cyw43_wifi()`.

In Meshtastic firmware we use the redefined `initVariant` function to reduce the clock speeds for power saving on the non-Wifi Pico boards. This redefinition will not work for the Wifi boards because of the above initialization of the CYW43.

Is there a workaround/suggestion to redefine the `initVariant` outside arduino-pico and call the `init_cyw43_wifi()` from there?
Or is it possible to have for example a preprocessor condition that disables the use of `initVariant` in arduino-pico for use with an external function?

earlephilhower, 2025-02-16:
That's an interesting request.  `initVariant` has to be a strong definition in the variant file or there would be no guarantee it would get linked in over the dummy `weak` version in `main.cpp`.

Are you trying to disable WiFi/BT (and LED GPIO) completely?  Or just have it set up with the proper lower SYSCLK?  If it's just a matter of getting the SYSCLK set before `initVariant` then the `runtime_init_xxx` infrastructure that's already there can do it.  Look at `psram.cpp` for the `PICO_RUNTIME_INIT_FUNC_RUNTIME` usage (or the plain SDK, that's how the chip is brought up before the C runtime).

Mictronics, 2025-02-16:
We use the strong definition in Meshtastic variant file to set up a lower SYSCLK as early as possible in the boot process so all the peripherals are init with the lower clock, e.g. UARTs. Wifi shall be enabled on the PicoW/Pico2W but init from the external ìnitVariant.

I will have a look on your `runtime_init_xx`suggestion and report back.

Mictronics, 2025-02-16:
Thanks Earl for the suggestion. `runtime_init_xx`is a solution.
Working on the RP2350 and I will move to that also on the RP2040 and free up the `initVariant` function.

closed 2025-02-16 by Mictronics