earlephilhower/arduino-pico #2806: I2S stops working after including SimpleFOC library

ZachMckennedyFWig, 2025-02-17:
I'll preface this by saying it is probably an issue with my code/understanding of how the two libraries work under the hood but I cannot figure out how to solve it for the life of me. 
---
I'm working on a project that requires audio playback and control of a motor at the same time on the RP2040. I've gotten a basic script that plays back audio from an SPI SD card reader through I2S to a DAC and a separate script that controls a motor over 6PWM channels and a SPI magnetic encoder. 

Independently the scripts work fine, but when I combine the code from both scripts together the DAC is being fed pure noise over I2S despite the buffers being fed into the `i2s.write();` command being identical to the ones in the I2S only script. 

Hardware being used:
- Raspberry Pi Pico
- HiLetgo 6 Pin Micro SD Card Reader (Attached to SPI0 on GPIO `16`, `17`, `18`, `19`)
- PCM5102 I2S DAC (Attached to `GPIO 2: DATA`, `GPIO 3: BCLK`, `GPIO 4: LRCK`)
- TMC6300 BLDC Motor Driver (PWM signals attached to GPIO `10: UH_PIN`, `11: UL_PIN`, `8: VH_PIN`, `9: VL_PIN`, `6: WH_PIN`, `7: WL_PIN`)
- AS5X47U SPI Magnetic encoder (Attached to SPI1 on GPIO `12`, `13`, `14`, `15`)

Here is the current code I am trying to get working: 
```C++
#include <Arduino.h>
#include <I2S.h>
#include <SdFat.h>
#include <SimpleFOC.h>
#include <SPI.h>
#include <algorithm>
#include <iterator>

extern "C" {
#include "hardware/dma.h"
#include "hardware/irq.h"
}

// Pins
#define SD_CS_PIN 17
#define I2S_BCLK_PIN 3
#define I2S_DATA_PIN 2

// Audio Variables:
const size_t bufferSize = 1024;
// Actual buffer length for 16 bit audio
const size_t buffer_len = bufferSize*2U;
uint8_t buffer[buffer_len];
volatile bool bufferReady = false;
SdFat sd;
FsFile wavFile;
int dmaChan;
I2S i2s(OUTPUT);

// Motor Variables:

// Phase U:
#define UH_PIN 10
#define UL_PIN 11
// Phase V:
#define VH_PIN 8
#define VL_PIN 9
// Phase W
#define WH_PIN 6
#define WL_PIN 7
// BLDC motor & driver instance
BLDCDriver6PWM driver = BLDCDriver6PWM(UH_PIN, UL_PIN, VH_PIN, VL_PIN, WH_PIN, WL_PIN);
// BLDCMotor( pp number , phase resistance, KV rating)
BLDCMotor motor = BLDCMotor(3, 3.6, 1920.0, 0.00018); 
// instance of AS5047/AS5147 sensor
#define SPI_1_MISO  12
#define SPI_1_MOSI  15
#define SPI_1_SCK   14
#define ENC_CS  13 
MagneticSensorSPI sensor = MagneticSensorSPI(AS5147_SPI, ENC_CS); 

// DMA callback for I2S audio
void onTX(){
    // Function called when 1 of the DMA buffers is fully transmitted. 
    bufferReady = true;
}

// Function to adjust the volume of the audio
void buf_volume(bool rev) {
    for (int i = 0; i < buffer_len; i += 2) {
        // Reassemble the sample from two bytes (little-endian)
        uint8_t lower = buffer[i];
        uint8_t upper = buffer[i+1];
        int16_t sample = ((int16_t)upper << 8) | lower;

        // Lower the volume by dividing the sample value.
        sample = sample / 64;

        // Split the sample back into two bytes.
        buffer[i] = (uint8_t)(sample & 0x00FF);       // Lower 8 bits.
        buffer[i+1] = (uint8_t)((sample >> 8) & 0x00FF);  // Upper 8 bits.

        // Reverse audio if passed
        if(rev) {std::reverse(std::begin(buffer), std::end(buffer));}
    }
}

void setup() {
    // https://arduino-pico.readthedocs.io/en/latest/i2s.html    <-- Docs for this PIO I2S library. 
    i2s.setFrequency(44100);
    i2s.setBitsPerSample(16);
    i2s.onTransmit(onTX);
    i2s.setDATA(I2S_DATA_PIN);
    i2s.setBCLK(I2S_BCLK_PIN); // Note: LRCLK = BCLK + 1
    i2s.setBuffers(4, bufferSize * sizeof(int16_t) / sizeof(uint32_t));
    i2s.begin();

    Serial.begin(115200);
    while (!Serial) {delay(100);};
 
    // Intialize the Encoder:
    SPI1.begin(ENC_CS);
    sensor.init(&SPI1);
    motor.linkSensor(&sensor);

    // driver config
    driver.voltage_power_supply = 5;
    driver.pwm_frequency = 40000;
    motor.linkDriver(&driver);

    // controller configuration 
    motor.controller = MotionControlType::velocity;
    motor.PID_velocity.P = 0.075;
    motor.PID_velocity.I = 1.0;
    motor.PID_velocity.D = 0.0001;
    motor.PID_velocity.output_ramp = 300;
    motor.LPF_velocity.Tf = 0.005;
    motor.current_limit = 1; // Amps
    motor.voltage_limit = 5.0;
    motor.foc_modulation = FOCModulationType::SpaceVectorPWM;

    // Skip Encoder Initialization
    motor.sensor_direction = Direction::CCW;
    motor.zero_electric_angle = 0.73; 

    // Init Simple FOC
    driver.init();
    motor.init();
    motor.initFOC(); 
    Serial.println("Motor ready!");

    // Initialize SD card with SdFat
    if (!sd.begin(SD_CS_PIN, SD_SCK_MHZ(10))) {  // 10 MHz SPI speed
        Serial.println("SD Card failed!");
        while (1);
    }
    // Attempt to open the file
    if (!wavFile.open("17.wav", O_READ)) {
        Serial.println("Failed to open 17.wav");
        return;
    }
    Serial.println("WAV OPENED!");

    // Skips header of the wav file
    wavFile.seek(44);
    // Fill all 4 DMA buffers with audio
    for(int i = 0; i < 4; i++) {
        wavFile.read(buffer, buffer_len);
        buf_volume(false);
        i2s.write(buffer, buffer_len);
    }
    bufferReady = false;
}

void loop() {
    motor.loopFOC();
    motor.move(10.0);

    // When buffer is transmitted, write a new one
    if(bufferReady) {
        wavFile.read(buffer, bufferSize*sizeof(int16_t));
        buf_volume(false);
        i2s.write(buffer, bufferSize*sizeof(int16_t));
        bufferReady = false;
    }
}
```

And here is my Platformio.ini:
```C++
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
lib_deps = 
	askuric/Simple FOC@^2.3.4
	greiman/SdFat@^2.3.0
build_flags = -D USE_SD_FAT
lib_archive = false
```

Potential Issues that I've ruled out:
- Not the PIO channels being filled, the Simple FOC library does not use PIO
- Not the simple FOC library changing the system clock speed after I2S is initialized

My guesses: 
- DMA channels are conflicting

Thank you in advance for any help/ideas for whats going on!

earlephilhower, 2025-02-17:
Since this is related to a 3rd party library, why not try there first?  I doubt many here have any experience with it and I personally haven't heard of it or have any of the HW needed to run BLDC motors.

In general, there's no good reason to pull in SdFat from grieman, it's already untouched and running in here as part of the core.  I'd recommend using the Arduino `File` support and `SD` instead of the low-level SdFat calls (but they'll work as long as you don't expect it to ever work with anything else like LittleFS or FsFAT).  FWIW we use the 2.3.0 release natively as well so this is probably a no-op.

The core uses the SDK to reserve DMA and PIOs, so if that's the real problem then the 3rd party library may need to use the SDK allocator and not assume certain blocks are available.

ZachMckennedyFWig, 2025-02-17:
> Since this is related to a 3rd party library, why not try there first?  I doubt many here have any experience with it and I personally haven't heard of it or have any of the HW needed to run BLDC [...]

Thanks for the reply,

I should have initially clarified that I started by reaching out to the simpleFOC devs through their discord and they had no idea why this issue would be happening. They suggested that it might be because the proximity of the PWM GPIO pins for the BLDC driver and the I2S pins because they could be "fussy" sometimes. They ultimately suggested reaching out to you guys to see if anyone had an idea of what could be going on though. 

I'll keep trying to debug and forward your suggestion to them. I'll also change the Sd code like you suggested.

Thank you again for your time.