earlephilhower/arduino-pico #2808: XIAO Rp2350 Wire1 not working.

metanav, 2025-02-18:
I am using a [XIAO RP2350](https://jp.seeedstudio.com/Seeed-XIAO-RP2350-p-5944.html) to connect a[ Grove - CO2 & Temperature & Humidity Sensor（SCD41）](https://jp.seeedstudio.com/Grove-CO2-Temperature-Humidity-Sensor-SCD41-p-5025.html) over I2C. This sensor works perfectly using Wire but when I use the Wire1 it does not work. I have done the changes below in the file `rp2040/hardware/rp2040/4.4.0/variants/seeed_xiao_rp2350`.

```
//#define PIN_WIRE1_SDA  (16u)
//#define PIN_WIRE1_SCL  (17u)
#define PIN_WIRE1_SDA  (0u)
#define PIN_WIRE1_SCL  (1u)
```

The I2C scan code recognizes the sensor address using either Wire or Wire1. I am using the GPIO6(SDA) and GPIO7(SCK) for Wire and GPIO0(SDA) and GPIO1(SCK) for Wire1. Just in case, I tried another I2C sensor but no success.

closed 2025-02-18 by earlephilhower

earlephilhower, 2025-02-18:
Actually, looking into the contributed Xaoi RP2350 pins_arduino.h, it's really messed up.  They've redefined the Wire device (but not Wire1, so both `Wire` and `Wire1` will try and use the same HW) and confused the wire0/1 pins.

The website shows
[image]
[image]

So no need to swap I2C devices at all.  I will update #2811 to move `Wire` to the sda0/scl0 16/17, remove the swapping of i2c devices, and set `Wire1` to sda1/scl1 on the 6/7 pinout.

metanav, 2025-02-19:
Thank you, @earlephilhower! Both Wire and Wire1 are working now!

commit 0b67b8fb (2025-02-18): Fix Wire/Wire1 definitions for Xaio RP2350

Fixes #2808
commit 3d17a56e (2025-02-18): Fix Wire/Wire1 definitions for Xaio RP2350 (#2811)

Fixes #2808