earlephilhower/arduino-pico #2817: pmalloc() issue on pimoroni PGA2350

rheslip, 2025-02-22:
Ported a fairly large sketch which uses pmalloc() and free() for all PSRAM variables from the Pimoroni Pico Plus 2 to their PGA 2350 board. Code works fine on the Pico Plus 2, fails on the PGA 2350. PGA2350 tested at various clock speeds and fails even at underclocked 100mhz.

A call to rp2040.getPSRAMSize() at the beginning of setup() correctly shows 8mb PSRAM. Immediately after that a call to rp2040.getFreePSRAMHeap() shows 0 bytes free.

A test sketch which pmallocs and frees 128kb chunks of PSRAM on the PGA 2350 works so it appears to have something to do with the larger sketch. I don't think this could be memory corruption because per above it fails in the first few lines of setup(). Perhaps a linker related issue?

earlephilhower, 2025-02-22:
I did a lot of development on the PGA2350, thanks to a kind user who got me one before they were readily available.  I've run the PSRAM example locally and it's running fine so you're going to need to provide an MCVE that shows the issue.

Note that the PGA2350 is actually a RP2350B with the extra GPIO pins, so if you have anything that touches I/O that could mess up a sketch (i.e. there are now 2 registers for GPIO state, drivers, etc.)

rheslip, 2025-02-22:
Thanks for the response. I think I may have a bad board. Sometimes it shows 8mb of heap, sometimes not. Other behavior has become inconsistent as well.

 Pico Arduino is awesome - thanks so much for your tireless work on this!

rheslip, 2025-02-23:
Issue resolved - I had assigned PSRAM chip select GPIO 47 as an I2S pin which caused the random PSRAM behavior.

earlephilhower, 2025-02-23:
Good debug!

closed 2025-02-23 by earlephilhower