earlephilhower/arduino-pico #2818: RP2350 PSRAM clock ignores core clock set in IDE.

dan2wik, 2025-02-23:
There appears to be issues when running PSRAM at non default frequencies.
I am running on 4.4.4, using a Pimoroni Pico Plus 2, and I am using the PSRAMTest example sketch.
The PSRAM is rated and programmed for 109MHz for this board.
I am testing the SCK speed using an oscilloscope for diagnostics.

I believe DDR is enabled for the following due to half expected clock being measured.
50MHz core, 109MHz PSRAM measures about 15MHz SCK and passes.
150MHz core, 109MHz PSRAM measures 50MHz SCK and passes.
200MHz core, 109MHz PSRAM measures about 67MHz SCK and fails.

150MHz core, 50MHz PSRAM measures about 25MHz SCK and passes.
200MHz core, 50MHz PSRAM measures about 33MHz SCK and passes.
300MHz core, 50MHz PSRAM does not give a PSRAM clock and fails.

In the following, I can only guess the speed drops low enough for DDR to disable.
50MHz core, 5MHz PSRAM measures about 1.7MHz SCK and passes.
150MHz core, 5MHz PSRAM measures 5MHz SCK and passes.
200MHz core, 5MHz PSRAM measures about 6.7MHz SCK and passes.
300MHz core, 5MHz PSRAM measures 10MHz SCK and passes.

From the data I collected, it seems the calculations are correct for 150MHz but the calculations do not scale to allow for changing of the core clock, creating stability issues at high clocks, and leaving performance on the table at low clocks.

earlephilhower, 2025-02-23:
Good catch.  The PSRAM clock is set [early in boot](https://github.com/earlephilhower/arduino-pico/blob/2f82bfd22a2e44f8c099f19d8cbf914be3e3d059/cores/rp2040/psram.cpp#L323) before `main` where the upgraded clocks are set.  It should be possible to call `set_psram_timing` when the sysclk is changed, same as we'll be doing with the vreg in the 2.1.(2?) SDK update to 200MHZ on the RP2040.

earlephilhower, 2025-02-24:
@dan2wik can you give #2824 a try and report back?  I don't have a reliable way of measuring the clockspeed of the QSPI interface.  The new code adds an adjustment to the clocking immediately after setting the new sysclk.  There are limitations on the divisors for the QSPI clock so it's not an exact match, but the best the HW can do.

dan2wik, 2025-02-26:
This doesn't seem to have worked. Something has changed though.
All tested at 109MHz psram.
150MHz core, measures either 75 or 67MHz (My scope is at it's limits here). Divisor of 2.
50MHz core, measures 25MHz. Divisor of 2. (Of note, if I force the divisor to 1, I can get 50MHz stable on psram)
200MHz core, I was unable to get a sensible measurement but PSRAM test fails so likely 100MHz with divisor of 2.
240MHz core, fails test and measures somewhere between 140 and 110MHz which would likely be 120MHz with the divisor still stuck at 2.

earlephilhower, 2025-02-26:
The clock logic really boils down to
https://github.com/earlephilhower/arduino-pico/blob/6236d1f7c599adf20c5b3a479e7232e2678c4fe2/cores/rp2040/psram.cpp#L202-L204

There is no DDR set via M1_WFMT.DTR
https://github.com/earlephilhower/arduino-pico/blob/6236d1f7c599adf20c5b3a479e7232e2678c4fe2/cores/rp2040/psram.cpp#L302-L309

Per the [datasheet](https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf) then it should just be an integer divider and the math for the default 150mhz core/109mhz psram gives `floor((150+109)/109) = 2` so you should be seeing 75MHz on the qspi clock pin in default operation.

The registers I get for default 150mhz operation here are
08:51:04.242 -> FMT 000012aa
08:51:04.242 -> TIM 61a48102

wfmt shows no DDR (bit 28==0)
timing shows divider = 2 == 150/2 = 75MHz  (divider is the lowest byte of the register)

You can dump the registers when over//underclocking by adding
````
#include <hardware/structs/qmi.h>
#include <hardware/structs/xip_ctrl.h>
````
and
````
Serial.printf("FMT %08x\n", qmi_hw->m[1].wfmt);
Serial.printf("TIM %08x\n", qmi_hw->m[1].timing);
````
in the loop of PSRAMTest (or anything when PSRAM is enabled in the board).

Running the code and checking some overclodk
At 150MHz it shows div==2 ==> 75mhz
Ar 50MHz it shows div==1 ==> 50MHz

So, the logic is setting the register properly (with the change in #2824) as far as I can see.  The datasheet says the clock can be adjusted at any time but for overclocking you should do a dummy read to ensure the interface clock is updated for the 1st real read after changing sysclk, which we aren't doing, but that's immaterial for the default and underclocking measured  clocks not matching calculations.

[image]

TLDR:  I think a more basic test needs to succeed on your system before we dig into anything.  150MHz should show the PSRAM QSPI at 75MHz.  If it doesn't, then either a) the measurement is wrong (wrong pin, noise, etc.) or b) the HW is wrong.  We've already found a HW bug in the cache clear logic for the XIP so (b) is definitely a possibility (but given the simplicity of seeing how fast PSRAM is running, unlikely)...

dan2wik, 2025-02-26:
In the last test I believe I did measure 75mhz psram at 150mhz core, so a divisor of 2 which is correct.
But for 50mhz I only measured 25mhz so divisor still 2, but I was also able to confirm I could get 50mhz at 50mhz by forcing a divisor of 1 in psram.cpp.

Tomorrow I'll read some register data and see if I can confirm the set divisors are relected in hardware in my instance, and I'll troubleshoot other factors more from there.

On a side note, my rp2350 doesn't seem to work at all when set to 175mhz. Blink sketch doesn't run, usb never negotiates. I only have 1 rp2350 board so I'm wondering if there's a hardware issue or if it's software. Please check and let me know.

earlephilhower, 2025-02-26:
Can you double check that you've got the patch applied?  At 50MHz the register shows it's already at div=1, no hacking to psram.cpp needed.  With `master` it will, however, be at div/2 (25mhz).

I haven't really looked into overclocking the 2350, and if it doesn't work that's still legit and not really a problem.  It's a completely different RTL than the 2040.  But underclocking we should be good PSRAM-clock-wise...

dan2wik, 2025-02-26:
I've re-installed my environment and I guess the patch never applied correctly on the old one. It's working well now.
I've tested all frequencies up to 300MHz.
50-100MHz runs clockdiv 1 correctly and stable.
120-200MHz runs clockdiv 2 correctly and stable. (175MHz doesn't boot at all)
225-250MHz runs clockdiv 3 correctly and stable.
275MHz doesn't boot at all.
300MHz fails test at clockdiv 3, but passes at clockdiv 4.
360MHz fails at clockdiv 4 and passes at clockdiv 5.
The QIO hardware likely just can't keep up being this far out of spec. (Noting down for my reference mostly)

PR #2824 (2025-02-27): Adjust the PSRAM clock when over/underclock F_SYS
Fixes #2818
commit 8deb6b97 (2025-02-27): Adjust the PSRAM clock when over/underclock F_SYS (#2824)

* Adjust the PSRAM clock when over/underclock F_SYS

Fixes #2818

* Need to increase PSRAM divider before sysclk increase

Per datasheet, when increasing sysclk speed we need to set the qmi
clocks first and do a dummy transfer to ensure no invalid speed
operations happen on the bus.  Handle the logic for this while setting
up the overclock.
closed 2025-02-27 by earlephilhower