earlephilhower/arduino-pico #2820: No support for Olimex PICO2-XL(2MB)/XXL(16MB)

hlide, 2025-02-23:
As the title says, I cannot see any reference to PICO2-XXL when trying to select board in Arduino Arduino IDE 2.x.x using version 4.4.4.

earlephilhower, 2025-02-23:
Since you seem to have one or both in hand, why not try a stab at [making a new board definition](https://github.com/earlephilhower/arduino-pico/blob/master/docs/contrib.rst#adding-a-new-board)?

OTW, you can use the `Generic RP2350` and adjust your flash and PSRAM config as appropriate.

hlide, 2025-02-23:
I wish but I'm not familiar at all with the board definition. 

I'll try the generic one. 

earlephilhower left a comment [ https://github.com/earlephilhower/arduino-pico/issues/2820#issuecomment-2677072870 | (earlephilhower/arduino-pico#2820) ]

PR #2868 (2025-03-18): Add Olimex Pico2XL and Pico2XXL
Fixes #2820
commit 50b9ea99 (2025-03-18): Add Olimex Pico2XL and Pico2XXL (#2868)

Fixes #2820
closed 2025-03-18 by earlephilhower