earlephilhower/arduino-pico #2832: Updating the RTC?

TheOracle2212, 2025-03-02:
Is there any simple way of accessing the Real Time Clock (RTC) on the Pi Pico? I'm building a GPS based clock and I'd like to update the RTC on the Pi Pico every time I get a valid GPS timestamp.

Thanks in advance!

maxgerhardt, 2025-03-02:
Can you check out this code?

```cpp
#include <Arduino.h>
#include <hardware/rtc.h>

void setup() {
  rtc_init();
  datetime_t date_to_set = {
    .year = 2025,
    .month = 3,
    .day = 2,
    .hour = 21,
    .min = 14,
    .sec = 30
  };
  rtc_set_datetime(&date_to_set);
}

void loop() {
  datetime_t cur_date;
  rtc_get_datetime(&cur_date);
  Serial.println("Current date: " + String(cur_date.year) + "/" + String(cur_date.month) + "/" + String(cur_date.day) 
    + " " + String(cur_date.hour) + ":" + String(cur_date.min) + ":" + String(cur_date.sec));
  delay(3000);
}
```

earlephilhower, 2025-03-02:
The physical RTC block on the RP2040 really isn't all that useful outside of waking the chip up after deep sleep, in my experience.  They even dropped it completely in the RP2350(Pico 2).  So if you just want to set the time proper for POSIX calls (i.e. `time()`) then the normal call works
https://github.com/earlephilhower/arduino-pico/blob/46a58fb4b5ffc7b62ca48c1451e5299ef8b97aa3/cores/rp2040/posix.cpp#L108

That requires a `timeval` struct to be built, but I think GPS is like NTP and gives fractional seconds since an epoch.  With that, you can use the internal function we use for LWIP SNTP which takes integer seconds and microseconds:
https://github.com/earlephilhower/arduino-pico/blob/46a58fb4b5ffc7b62ca48c1451e5299ef8b97aa3/cores/rp2040/posix.cpp#L121

TheOracle2212, 2025-03-03:
Thanks for your replies. I managed to use @maxgerhardt's program and sync the inbuilt RTC to the GPS whenever the time was valid. 

> but I think GPS is like NTP and gives fractional seconds since an epoch.
Not really, it just spits out the NMEA sentences continuously which include a DD:MM:YYYY and HH:MM:SS (and in some cases CCCC) timestamp.

> They even dropped it completely in the RP2350(Pico 2)
Why? Not used as much as they thought? I can see some really good uses for it instead of springing for an external RTC?

(What does POSIX mean??)

earlephilhower, 2025-03-03:
POSIX as in the standardized library calls.  i.e. `time`, `asctime`, and `localtime` among others.  The core uses the internal clock set via the `settimeofday` system call, not the RTC, for timekeeping.  If you manage it yourself, feel free to use the RTC instead (but it's not there if you move to the Pico 2)...

closed 2025-03-03 by earlephilhower