earlephilhower/arduino-pico #2835: SerialUART setRX/setTX pins missing RP2350 GPIO function select F11 pins

deltaford, 2025-03-03:
RP2350 allows for more GPIO pins to be used for RX/TX as defined in GPIO function select F11 and seem to be missing in the current definition.

`bool SerialUART::setRX(pin_size_t pin) {
#ifdef PICO_RP2350B
    constexpr uint64_t valid[2] = { __bitset({1, 13, 17, 29, 33, 45}) /* UART0 */,
                                    __bitset({5, 9, 21, 25, 37, 41})  /* UART1 */

bool SerialUART::setTX(pin_size_t pin) {
#ifdef PICO_RP2350B
    constexpr uint64_t valid[2] = { __bitset({0, 12, 16, 28, 32, 44}) /* UART0 */,
                                    __bitset({4, 8, 20, 24, 36, 40})  /* UART1 */`

earlephilhower, 2025-03-03:
Yes, looks like those got missed,  Since you found it and probably have a need for it, want to make a PR?

deltaford, 2025-03-04:
If it was only the definition that need updating i could make PR for it.

```
bool SerialUART::setRX(pin_size_t pin) {
#if defined(PICO_RP2350B)
    constexpr uint64_t valid[2] = { __bitset({1, 3, 13, 15, 17, 19, 29, 31, 33, 35, 45, 47}) /* UART0 */,
                                    __bitset({5, 7, 9, 11, 21, 23, 25, 27, 37, 39, 41, 43})  /* UART1 */
                                  };
#elif defined(PICO_RP2350A)
    constexpr uint64_t valid[2] = { __bitset({1, 3, 13, 15, 17, 19, 29})     /* UART0 */,
                                    __bitset({5, 7, 9, 11, 21, 23, 25, 27})  /* UART1 */
                                  };
#else
    constexpr uint64_t valid[2] = { __bitset({1, 13, 17, 29}) /* UART0 */,
                                    __bitset({5, 9, 21, 25})  /* UART1 */
                                  };
#endif

```
...
```
bool SerialUART::setTX(pin_size_t pin) {
#if defined(PICO_RP2350B)
    constexpr uint64_t valid[2] = { __bitset({0, 2, 12, 14, 16, 18, 28, 30, 32, 34, 44, 46}) /* UART0 */,
                                    __bitset({4, 6, 8, 10, 20, 22, 24, 26, 36, 38, 40, 42})  /* UART1 */
                                  };
#elif defined(PICO_RP2350A)
    constexpr uint64_t valid[2] = { __bitset({0, 2, 12, 14, 16, 18, 28})     /* UART0 */,
                                    __bitset({4, 6, 8, 10, 20, 22, 24, 26})  /* UART1 */
                                  };
#else
    constexpr uint64_t valid[2] = { __bitset({0, 12, 16, 28}) /* UART0 */,
                                    __bitset({4, 8, 20, 24})  /* UART1 */
                                  };
#endif
```
....

But after looking closer it seem it would also require changes to SerialUART::begin, changes beyond my knowledge of this core and the pico-sdk.
Currently TX/RX GPIO is always initialized as GPIO function GPIO_FUNC_UART but for F11 it would have to be GPIO_FUNC_UART_AUX.

```
    gpio_set_function(_tx, GPIO_FUNC_UART);
    gpio_set_outover(_tx, _invertTX ? 1 : 0);
    gpio_set_function(_rx, GPIO_FUNC_UART);
    gpio_set_inover(_rx, _invertRX ? 1 : 0);
```

SerialUART::begin would need to somehow consider both GPIO_FUNC_UART & GPIO_FUNC_UART_AUX for rp2350

earlephilhower, 2025-03-04:
No worries.  I did the simple brute force way of checking in #2837.  Can you give that a try on your setup and report back?

deltaford, 2025-03-05:
Successfully tested your PR using both UART configured on GPIO_FUNC_UART_AUX pins. Thanks!

commit 1130198f (2025-03-04): Enable add'l UART_AUX pinouts for RP2350

Fixes #2835.  Thanks @deltaford!
PR #2837 (2025-03-05): Enable add'l UART_AUX pinouts for RP2350
Fixes #2835.  Thanks @deltaford~