earlephilhower/arduino-pico #2838: Per @maxgerhardt this is [still WIP](https://github.com/earlephilhower/arduino-pico/pull/2620#issuecomment-2509130760).  He does work on lots of different embedded systems so probably got caught up with other things recently.

seamusdemora, 2025-03-04:
Per @maxgerhardt this is [still WIP](https://github.com/earlephilhower/arduino-pico/pull/2620#issuecomment-2509130760).  He does work on lots of different embedded systems so probably got caught up with other things recently.

The Mac CI issue is a non-issue, related to GH sunsetting older Mac system images which CI called.  The next time Max merges with `master` the CI will re-run with the updated Mac x86 runner.

_Originally posted by @earlephilhower in https://github.com/earlephilhower/arduino-pico/issues/2620#issuecomment-2573991026_

seamusdemora, 2025-03-04:
@earlephilhower : I thought I'd try opening this as a new issue as all the others I was following on this subject have been closed.  Apologies if this was inappropriate - you may close this also...  I'll "get the message"  :) 

Anyway - it's been a while since there has been any word on this low power library that @maxgerhardt was working on.  I'm still very keen to try it, but I thought I should "raise a flag", and ask if it's still forthcoming - or if it's been abandoned.

Thanks for any information.

earlephilhower, 2025-03-18:
#2528 would be where to track this.  Looks like he's occupied elsewhere.  He's got some code in the PR already, so why not take it as a starting point?  I unfortunately don't have a good setup to measure currents that low so can't really work much on it...

closed 2025-03-18 by earlephilhower

seamusdemora, 2025-03-19:
> [#2528](https://github.com/earlephilhower/arduino-pico/issues/2528) would be where to track this. Looks like he's occupied elsewhere. He's got some code in the PR already, so why not take it as a [...]

I would have done that already if I could have gotten an answer to my [last question for @maxgerhardt](https://github.com/earlephilhower/arduino-pico/issues/2528#issuecomment-2471767760).  Can you direct me to something that provides any sort of documentation on how to use a library that (AFAIK) hasn't actually been committed?  Sorry - but I'm not an experienced user of the Arduino system, and it's been months since my last usage. 

And I'll live without the current measurement for now... I've got an inline USB V&I monitor. 

Thanks for your time.

earlephilhower, 2025-03-19:
If you don't like `git then you can `wget https://github.com/earlephilhower/arduino-pico/pull/2620.patch` and then `patch -p1 < 2620.patch` in your copy of the core.  OTW you can use the normal technique described in the GH docs https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/reviewing-changes-in-pull-requests/checking-out-pull-requests-locally