earlephilhower/arduino-pico #2846: EEPROM, uf2 and elf difference

VitRuzicka, 2025-03-10:
Hi, 
I came across an interesting issue. I have a program that uses EEPROM emulation on the rp2040. Now the issue:
- When flashing the board from vscode (uploading elf with picotool), the issue is not present. Settings in EEPROM are kept.
- When flashing by holding BOOTSEL and dragging uf2, the issue is present. Settings in EEPROM are kept.

The issue is that the board has also ethernet chip on SPI which stops communicating, probably because some of the old data of the old firmware stays in flash (EEPROM) when the new firmware is uploaded.  I deduced this, because when I first upload flash_nuke.uf2 and then the new firmware, the Ethernet works as expected (the old data in flash is completely erased)

I wanted to know, if there is (obviously there is) any difference when flashing by elf or by uploading uf2 file?

Thanks, 
Vit

earlephilhower, 2025-03-11:
Without a small MCVE that can run without special HW, there's not much we can do here.  Please supply one or we're really not able to do much.

In general there's no difference using picotool (or openocd) vs. UF2, as they both just interpret specific sections of the ELF and write them to predefined flash sectors.  EEPROM wouldn't be erased on the rp2040 when an app is uploaded so if you're doing something like storing naked structs in it, you'll need to carefully version them.  Or, better yet, stop using EEPROM and just use LittleFS and a small file (easy to `write` binary or even print human-readable text values).  For picotool and UF2 the reboot is identical (but for openocd, the reboot after upload will be a slightly different method).

VitRuzicka, 2025-03-11:
Hi, 
thank you for the tip about the LittleFS, it seems a lot cleaner. I will give it a shot. 
The issue overall doesnt seem to be related to the EEPROM as I wrote earlier, but rather a network switch issue.
Thank you for your time and also for writing such awesome toolkit.