earlephilhower/arduino-pico #2850: Problems with SPI / Unable to assign pins

Zyga05, 2025-03-12:
Hello, 
I have a problem using SPI interface. I would like to display some information on LCD display using U8G2 library. Software SPI works, however problem starts with hardwere SPI. I have been trying to initialize SPI1 interface, but no progress has been made. I am using my custom PCB, which was designed to work only with SPI1 (for my convenience). PCB is working fine, I can comunicate with the IC, turn on some diodes etc. Most importantly, the connection between hardware SPI and display is correct so I guess this is not the PCB fault. Here is my code: 

`
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_ST7567_ENH_DG128064I_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 12, /* dc=*/ 13, /* reset=*/ 14); // Standard constructor for U8G2

void setup() {
  SPI1.setSCK(10);
  delay(1000);
  SPI1.setTX(11);
  delay(1000);
  SPI1.begin();
  delay(1000);
  u8g2.begin();
  delay(1000);
  pinMode(7, OUTPUT); // For backlight
  analogWrite(7, 150); // For backlight
  u8g2.setContrast(170);
}

void loop() {
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.drawStr(0,10,"Hello World!");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
  delay(1000);  
}

`

I read that U8G2 uses standard spi.h library and gets hardware information from there, but even after manipulating spi.h file to forcefully assign pins, no signs of life were shown. If I am missing something, please let me know.

dquadros, 2025-03-12:
IIRC, U8G2 uses only the default SPI (SPI0) .  You will have to change all SPI references in U8G2 to SPI1 to get it working. Another way (that will affect all your RP2040/RP2350 apps) is to change the definition of SPI and SPI1 in SPI.cpp.

The best option would be for the maintainers of the U8G2 to change the library to allow to specify the SPI object that will be used. This would be useful for microcontrollers with more than one SPI.

lyusupov, 2025-03-12:
[image]

[image]

[image]

[image]

earlephilhower, 2025-03-12:
Closing since this isn't a core issue.  When setting the pinouts, there it no need for `delay` between `setRX/TX` or before `begin`.  But do be aware that only certain pins are legal for SPI0 and SPI1, as documented in the RPI datasheet.  For other pinouts, look at `SoftwareSPI` here which doesn't use the HW and can run on any set of consecutive pins, mostly.

closed 2025-03-12 by earlephilhower