earlephilhower/arduino-pico #2851: Cannot use pinMode() with noInterrupts()

james1236, 2025-03-13:
```cpp
void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println("serial is working"); //#1
  delay(1000);

  noInterrupts();
  pinMode(3, INPUT); //comment/uncomment 
  interrupts();

  Serial.println("setup end"); //#2
}

void loop() {
  Serial.println("loop"); //#3
  delay(1000);
}
```

When I disable interrupts using `noInterrupts()` and then call `pinMode()`, the first print statement works (`serial is working`), but the subsequent ones (`setup end` and `loop`) do not.

I haven't yet confirmed whether this causes a crash or a hang as I currently don't have access to my debug probe.

Interestingly, this behavior doesn't occur when using an Arduino Uno in the Wokwi online simulator, so I suspect it might be specific to arduino-pico. 

Is it expected behavior for `pinMode()` to malfunction when interrupts are disabled? I’ve noticed this being done in a library I'm using, but I’m unsure if it’s considered safe and reliable.

earlephilhower, 2025-03-13:
Thanks for the MCVE but I can't reproduce a problem.  I'm using the latest 4.5.0 core and both a PicoW and Pico.

There is no restriction on IRQs and `pinMode` in the code. https://github.com/earlephilhower/arduino-pico/blob/e60858c327690222594a8b35bc65f68e52491c5c/cores/rp2040/wiring_digital.cpp#L29-L85

Are you sure you're using this core and not the Arduino.cc MBED-OS one?  They might need interrupts for some of the OS-level work of assigning a pin direction.  But here, we really don't care.

james1236, 2025-03-13:
Can confirm this works fine on `FQBN: rp2040:rp2040:rpipico`, I was accidentally using `FQBN: arduino:mbed_rp2040:pico`. Sorry.

closed 2025-03-13 by james1236