earlephilhower/arduino-pico #2856: Feature Request: Add Waveshare RP2350-ETH

pavleknezevic, 2025-03-14:
Board name: "RP2350-ETH"

Additional information:
https://www.waveshare.com/wiki/RP2350-ETH
https://files.waveshare.com/wiki/RP2350-ETH/RP2350-ETH%20Schematic%20diagram.pdf

earlephilhower, 2025-03-14:
Looks like a bog standard Pico2, even Dane flash size.  You should try the Pico2 or Generic RP2350 in the boards menu.

I've never seen that chip before, but reading the datasheet makes it clear you won't be using any of the network things here like MDNS or WebServer.  It has no support for anything but their limited single connection that I can see.  Definitely doesn't use LWIP.

For real networking, consider the Wiznet boards.  They have full networking stack integration here and are as powerful as the WiFi chips.

pavleknezevic, 2025-03-15:
I thought I was using it wrong.

I will be looking into the Wiznet boards.

Thank you!

closed 2025-03-15 by pavleknezevic