earlephilhower/arduino-pico #2879: Fatal Python error: initfsencoding: unable to load the file system codec

binarweb, 2025-03-21:
IDE: Arduino 2.3.4
OS: Windows 10 x64
Prev package version: 4.4.4
Current package version: 4.5.1
Additional Board Manager URL: https://github.com/earlephilhower/arduino-pico/releases/download/4.5.1/package_rp2040_index.json

Using `earlephilhower/arduino-pico` version 4.4.4 on Arduino IDE, everything worked fine.
Now, after I updated the package from Boards Manager, when I hit compile, I get:

```
Fatal Python error: initfsencoding: unable to load the file system codec
ModuleNotFoundError: No module named 'encodings'

Current thread 0x0000275c (most recent call first):

exit status 0xc0000409

Compilation error: exit status 0xc0000409
```

binarweb, 2025-03-21:
I fixed it in a non-traditional way.
I already had Arduino 1.8 installed. And at the time of the install it was installed before Arduino 2.3.

What I did:

1. uninstalled Arduino 2.3 and manually removed all folders I could find for this Arduino version in the Users \ AppData folder, both Local and Roaming.
2. installed again Arduino 2.3. But when I opened it, it still had ALL of my previous settings, including the Additional Boards URLs.
3. I opened Arduino 1.8 and added the URL to Additional Boards URLs, installed the latest package available, version 4.5.0, but there was no Olimex Pico2XXL, as it was not yet available in this package version.
4. I opened Arduino 2.3, chose a blink example and it worked!

I really do not understand what could have been the problem.

earlephilhower, 2025-03-21:
The issue is that sometimes the IDE doesn't kill the python executable before trying to uninstall/reinstall the package.  Release 4.5.1 and #2853 may help avoid this in the future, but it would only affect the future releases (4.5.2++), not upgrading from prior to 4.5.1.  

If you check the closed issues here for `encodings` you'll find the common solution is to exit the ide, maybe log out/log in to ensure everything is killed, manually deleting the install directory, and then reinstalling the latest release.

closed 2025-03-21 by earlephilhower

binarweb, 2025-04-15:
I uninstalled Python and both Arduino versions 1 and 2 (that I had installed), then cleaned up of all the traces by searching "arduino" and "python", reinstalled all 3 again and now it's working.

Thank you!