earlephilhower/arduino-pico #2881: Firmware RP2350 using UART Boot

Fantonrko, 2025-03-23:
Hi, I finally got my hands on the RP2350, I want to try to flash it via UART Boot, this feature was added in new versions, but I don't quite understand how to do it.
I wrote a code for ESP32 which contains a .bin file and writes it to the RP2350. I managed to start the UART mode using SS and SD1 pins, but the subsequent loading of data blocks does not lead me to the result, the RP2350 just stops responding (as indicated in the datasheet) but does not write (to flash)/start the firmware. I read that it is necessary to properly compile the .bin file, but I don't quite understand how to do it.

I am using Arduino IDE. Can anyone know how to solve this problem or tell me what I did wrong ? 

Translated with DeepL.com (free version)

earlephilhower, 2025-03-23:
UART boot doesn't take a standard bin or uf2.  You need to speak the proper protocol and you can only upload SRAM binaries which are not supported here.

closed 2025-03-23 by earlephilhower

Fantonrko, 2025-03-23:
> UART boot doesn't take a standard bin or uf2. You need to speak the proper protocol and you can only upload SRAM binaries which are not supported here.

Could you point me in the direction I need to go?
The datasheet says this:
```A UART boot binary is a normal RAM binary. It must have a valid IMAGE_DEF in order for the boot path to recognize it as a
bootable binary. The search window for the IMAGE_DEF is the whole of SRAM, but it’s recommended to place it close to
the beginning, because the bootrom searches linearly forward for the beginning of the IMAGE_DEF.```

earlephilhower, 2025-03-23:
Use the raw SDK and the SRAM build option, not this core.  I think you just upload the whole generated BIN and the ROM will find the header per the docs. The RPI forum can be of better assistance.

Fantonrko, 2025-03-23:
Okay, thanks!