earlephilhower/arduino-pico #2890: Use of pioasm in IDE build

greiman, 2025-03-30:
Is there any possibility of pioasm being available for IDE builds?

It is becoming difficult to provide a *.pio.h file in libraries.

Also as the sdk evolves the results from pioasm change.

Here are three stages of the  `struct pio_program.`

All had the same *.pio input file.

From an early [pioasm](https://wokwi.com/tools/pioasm): 

Has a totally different struct:
```
#define rd_data_wrap_target 1
#define rd_data_wrap 3

#define rd_data_offset_wait0 1u
#define rd_data_offset_wait1 2u

static const uint16_t rd_data_program_instructions[] = {
    0x20c7, //  0: wait   1 irq, 7                   
            //     .wrap_target
    0x2000, //  1: wait   0 gpio, 0                  
    0x2080, //  2: wait   1 gpio, 0                  
    0x4004, //  3: in     pins, 4                    
            //     .wrap
};

#if !PICO_NO_HARDWARE
static const struct pio_program rd_data_program = {
    .instructions = rd_data_program_instructions,
    .length = 4,
    .origin = -1,
};
```

The current version in this repo:

Note: ` .pio_version = 0,`

```
#define rd_data_wrap_target 1
#define rd_data_wrap 3
#define rd_data_pio_version 0

#define rd_data_offset_wait0 1u
#define rd_data_offset_wait1 2u

static const uint16_t rd_data_program_instructions[] = {
    0x20c7, //  0: wait   1 irq, 7                   
            //     .wrap_target
    0x2000, //  1: wait   0 gpio, 0                  
    0x2080, //  2: wait   1 gpio, 0                  
    0x4004, //  3: in     pins, 4                    
            //     .wrap
};

#if !PICO_NO_HARDWARE
static const struct pio_program rd_data_program = {
    .instructions = rd_data_program_instructions,
    .length = 4,
    .origin = -1,
    .pio_version = 0,
#if PICO_PIO_VERSION > 0
    .used_gpio_ranges = 0x1
#endif
};
```

The latest pioasm from Raspberry Pi:

Note:  `.pio_version = rd_data_pio_version,`
```
#define rd_data_wrap_target 1
#define rd_data_wrap 3
#define rd_data_pio_version 0

#define rd_data_offset_wait0 1u
#define rd_data_offset_wait1 2u

static const uint16_t rd_data_program_instructions[] = {
    0x20c7, //  0: wait   1 irq, 7
            //     .wrap_target
    0x2000, //  1: wait   0 gpio, 0
    0x2080, //  2: wait   1 gpio, 0
    0x4004, //  3: in     pins, 4
            //     .wrap
};

#if !PICO_NO_HARDWARE
static const struct pio_program rd_data_program = {
    .instructions = rd_data_program_instructions,
    .length = 4,
    .origin = -1,
    .pio_version = rd_data_pio_version,
#if PICO_PIO_VERSION > 0
    .used_gpio_ranges = 0x1
#endif
};
```

earlephilhower, 2025-04-01:
I'm traveling so can't really look into this, but do you have examples from other guess for something similar?  The Arduino build system really doesn't know how to handle anything other than C and C++ files.  A generic pre-build rule might be possible, calling something in Python that would scan the directory tree.  But that's probably a bad idea, making a new file after the build has been scheduled...

closed 2025-04-16 by earlephilhower

greiman, 2025-04-17:
Thanks for looking at the problem.

I will be working on a new version of PIO SDIO for the QFN-80 (RP2350B).  PIO can only access a limited range of  GPIOs on this chip.  There is new GPIOBASE register.  Also I will consider using new features in RP2350 PIO.

  I will try to protect from modifications of the C-SDK while working on the new version.