earlephilhower/arduino-pico #2901: Wrong PWM frequency if clk_sys frequency was changed

reald, 2025-04-10:
If the same pwm frequency is requested `analogWriteFreq( freq )` multiple times, the function does not request a reinitialization:

 ```
extern "C" void analogWriteFreq(uint32_t freq) {
    if (freq == analogFreq) {
        return;
 ```

However the sys clock frequency might have been changed  in mean time. 

`analogWrite()` does not check this and outputs a wrong PWM frequency then.

=> Add check for clk_sys and reinit PWM if clk_sys was changed.

earlephilhower, 2025-04-10:
This is not a normal operating mode and there's nothing in the core that actively changes the CPU frequency.  If you'd like to do a PR, I'd be happy to give it a look.

There are also other places where unexpected `sys_clk` changes will give you unexpected results, most likely.  Things like PIO-driven `tone` and the `ADCInput` class, for example.

earlephilhower, 2025-04-10:
Oh, and as a workaround, after doing a change to cyc_clk, you can `analogWriteFreq(NEEDEDFREQ + 1); analogWriteFreq(NEEDEDFREQ);` to force it to recalculate things.

closed 2025-05-05 by earlephilhower