earlephilhower/arduino-pico #2903: W55RP20 Ethernet support

ITstreet1, 2025-04-13:
Hello,

I just got my hands on this EVB. Tried the classic Blink sketch, it works. Ethernet, however, I didn't manage to make it work. 
Which package version should I try, and what example? Should I have to tweak a thing or two? Is there Ethernet support for this chip at all?

The earlier issue is closed.
https://github.com/earlephilhower/arduino-pico/issues/2492

earlephilhower, 2025-04-13:
First of all, what exact board definition did you have selected and what exact example (and/or changes made) did you run?  Every Ethernet/WiFi interface needs a specific driver to be used.

You need to select the W55RP20 board and use the https://github.com/earlephilhower/arduino-pico/blob/master/libraries/lwIP_w55rp20/examples/WiFiClient-W55RP20/WiFiClient-W55RP20.ino example (well, instantiate that `Wiznet55rp20lwIP` object)

The W55RP20 driver was submitted by WizNet themselves, but I personally don't have a model.  That said, I'd imagine they tested and got things running since they were good enough to give a serious chunk of driver code and that example.

ITstreet1, 2025-04-13:
I did select the WIZnet W55RP20-EVB-Pico from the Boards options. The example I used is exactly this one you mentioned: Examples -> IwIP_w55rp20 -> WiFiClient-W55RP20. I didn't make any changes to the code, except a CS change to 17:
Wiznet55rp20lwIP eth(17 /* chip select */);

The error I get is:
No wired Ethernet hardware detected. Check pinouts, wiring.

earlephilhower, 2025-04-13:
It's probably a pinout issue.  Where do you see the device has CS on pin 17?  Did you try their example which has it set to 1 by default?  A completely different and incompatible board, WIZNET_5100S_EVB_PICO, has CS=17 (and it's set to that in that particular example).

Looking at the schematic on their website, I can't really tell how they're wiring up.  Looks like an MCM/stack with an RP2040 die buried inside another device, since the RP2040 chip itself ships as a QFN56 and their board shows a single LQFP68 with direct magnetics connections.

If the example with CS=1 doesn't work, I suggest emailing them directly to get the proper wiring config. We can then update here since they do link to this project and did contribute the code themselves...

ITstreet1, 2025-04-13:
According to this:

https://docs.wiznet.io/Product/ioNIC/W55RP20/w55rp20-evb-pico

It says GPIO17, GPIO20, GPIO21, GPIO22, GPIO23, GPIO24, and GPIO25 are used internally. And as it uses the SPI (SPI0), then the GPIO17 is the CS. So 17 is just my guess. I tried the GPIO1, but got the same. The GPIO1 is available, so I doubt it is used internally.

The WIZnet W5500 EVB Pico is, I believe, the same as this one. Except, there are 2 chips. Just my guess. And according to this, it uses GPIO17 for CS, too.

https://docs.wiznet.io/Product/iEthernet/W5500/w5500-evb-pico

I will try to contact WIZnet.

maxgerhardt, 2025-04-14:
The way I read the datasheet and diagrams, especially the official example code

https://github.com/WIZnet-ioNIC/WIZnet-PICO-C/blob/3f9d8a13eae75aed2aaf18fa8d74ed98c90a3015/port/ioLibrary_Driver/inc/w5x00_spi.h#L18-L27

It seems to me that
```
SCLK:  GP21
MOSI: GP23
MISO: GP22
CS: GP20
RST: GP25
IRQ: GP24
```
for the W55RP20_EVB_PICO board. This also mirrors the code submitted by Wiznet

https://github.com/earlephilhower/arduino-pico/blob/e05dd50d626d2871ab988d8e72aa7c12e18e603c/libraries/lwIP_w55rp20/src/utility/w55rp20.h#L43-L46

So, does the default constructor
```cpp
Wiznet55rp20lwIP eth;
```
not work?

earlephilhower, 2025-04-14:
FWIW, the code seems to ignore the actual CS pin and SPI instance passed it.  `_cs` is set in the constructor but never used, only that `#define`s you mentioned.  They implement their own PIO-based SPI interface for comms to the wired Ethernet chip (those pins are not legal for any HW SPI by my quick inspection).

ITstreet1, 2025-04-15:
> The way I read the datasheet and diagrams, especially the official example code
> 
> https://github.com/WIZnet-ioNIC/WIZnet-PICO-C/blob/3f9d8a13eae75aed2aaf18fa8d74ed98c90a3015/port/ioLibrary_Driver/inc/w5x00_spi.h#L18-L27
> 
> It seems to me that
> 
> ```
> SCLK: GP21
> MOSI: GP23
> MISO: GP22
> CS:   GP20
> RST: GP25
> IRQ: GP24
> ```
> 
> for the W55RP20_EVB_PICO board. This also mirrors the code submitted by Wiznet
> 
> [arduino-pico/libraries/lwIP_w55rp20/src/utility/w55rp20.h](https://github.com/earlephilhower/arduino-pico/blob/e05dd50d626d2871ab988d8e72aa7c12e18e603c/libraries/lwIP_w55rp20/src/utility/w55rp20.h#L43-L46)
> 
> Lines 43 to 46 in [e05dd50](/earlephilhower/arduino-pico/commit/e05dd50d626d2871ab988d8e72aa7c12e18e603c)
>  #define WIZNET_PIO_SPI_MOSI_PIN 23 
>  #define WIZNET_PIO_SPI_MISO_PIN 22 
>  #define WIZNET_PIO_SPI_SCK_PIN  21 
>  #define WIZNET_PIO_SPI_CS_PIN   20 
> 
> So, does the default constructor
> 
> Wiznet55rp20lwIP eth;
> 
> not work?

I missed this one. After I changed CS (Wiznet55rp20lwIP eth) to 20, it worked. So, I think this CS from 1 had to be changed to 20 in your example.
If you need me to check a thing or two with this board, just let me know.

ITstreet1, 2025-04-15:
As I said, would you like me to check a thing or two on this board, or to close this issue?

earlephilhower, 2025-04-15:
Thanks, but I think we're good.  The IP infrastructure is the same for all networking here, so if the physical interface works (as it seems to be) then we're golden!

PR #2906 (2025-04-15): Update W55RP20 example for proper CS pin
Fixes #2903