earlephilhower/arduino-pico #2914: rp2040.getCycleCount64() does not work correctly on core 1

Celppu, 2025-04-17:
I try to read pwm using core 1 as the core 0 is busy with other tasks. I have simple task entry and pwm isr on core 1. I ty to get cyclecount but it seem to jump 24 bits at a time

`
PWM_ISR_1 GPIO: 17 Events: 12 High Time: 0 Low Time: 0 Last Time:  Time: 6358564865
PWM_ISR_1 GPIO: 17 Events: 12 High Time: 0 Low Time: 0 Last Time:  Time: 6358564865
PWM_ISR_1 GPIO: 17 Events: 12 High Time: 16777216 Low Time: 0 Last Time:  Time: 6375342081
PWM_ISR_1 GPIO: 17 Events: 12 High Time: 16777216 Low Time: 0 Last Time:  Time: 6375342081
PWM_ISR_1 GPIO: 17 Events: 12 High Time: 0 Low Time: 0 Last Time:  Time: 6375342081
PWM_ISR_1 GPIO: 17 Events: 12 High Time: 0 Low Time: 0 Last Time:  Time: 6375342081
`

How to get cyclecount on core 1? If I understand correctly there is seperate 24-bit timer and epoch?

earlephilhower, 2025-04-17:
Please provide a full, small MCVE that shows the issue.  Right now there's not enough info to look into anything, unfortunately.  FWIW the "cycles" here are intended to be sys_clk and they are the same core0/core1.  There is only 1 systick unit and the epoch is stored in a global value which is accessible from all cores.

If you are manually changing `systick_hw` in any way, you cannot use the core's implementation since it is dependent on that.  Look at the PIO based counter, then, as used when under FreeRTOS.  It burns a PIO SM but runs no matter what.

earlephilhower, 2025-04-17:
Oops, mybad.  It does seem each ARM core has its own systick timer.  I was under the impression that it was per-MCU but it's actually per-core per 2.4.12 of the datasheet...
````
2.4.1.2. Configuration
Each processor is configured with the following features:
....
• SysTick timer
````

So yes, the logic won't work properly since the counter isn't set up on the 2nd core when `setup1()` is launched.

earlephilhower, 2025-04-17:
Can you give #2915 a try and report back?  My simple test seems to work with it, with core 0 and core 1 both reporting monotonically increasing cycles:
````
void setup() {
}

void loop() {
  Serial.printf("0: %llu\n", rp2040.getCycleCount64());
  delay(100);
}

void setup1() {
}

void loop1() {
  Serial.printf("1: %llu\n", rp2040.getCycleCount64());
  delay(100);
}
````

Celppu, 2025-04-17:
Thank you for quick answer! I will get back to this next week to try out! I previously used

```
multicore_launch_core1(core1_entry);
```

But is there now difference how do I start core 1 to get the timer going?

PR #2915 (2025-04-18): Fix rp2040.getCycleCount() from core1
Fixes #2914

There are 2 systick units, one per core.  Set up and start core1's systick unit and track each core's epoch separately.
commit bc5b2c24 (2025-04-18): Fix rp2040.getCycleCount() from core1 (#2915)

Fixes #2914

There are 2 systick units, one per core.  Set up and start core1's
systick unit and track each core's epoch separately.

Document a method of preserving 100% user-only code on core1
and add a core1_disable_systick boolean flag that works like the
separate stack one.
closed 2025-04-18 by earlephilhower