earlephilhower/arduino-pico #2917: #define EARLE_PHILHOWER

PizzaProgram, 2025-04-18:
I'm humbelly asking to add a specific constant in the next release, that tells us, if the user has loaded **this** library. Something like:
```c
#define EARLE_PHILHOWER
// or a shorter version
#define EARLE_LIB
// or
#define PHILHOWER_LIB
```
... or similar. You decide, anything will do fine! _(except RP2040 or such, because it's too common, easy to mix up)_

I know, there is the "official" version:
```c
!defined(__MBED__)
```
but nobody can tell, what will happen in 5-10-15 years, when there will be still RPi chips out there, but maybe the 3th library appears.

I would like to finish my example code at Firmata, and I really need a solution which will work in the future too.

JAndrassy, 2025-04-18:
I would see this platform as the default. Any future alternative platform should maintain compatibility. 

and maybe use `#if defined(ARDUINO_ARCH_MBED)` not `__MBED__`

earlephilhower, 2025-04-18:
If the [recommended way](https://github.com/earlephilhower/arduino-pico/blob/master/docs/contrib.rst#compiler-defines-for-porting) isn't enough for you (and, really, it's been good enough for years of other people's use in hundreds of projects and libraries) you can use the version string that's part of the package process.
````
#ifdef ARDUINO_PICO_MAJOR
...
#endif
````

closed 2025-04-18 by earlephilhower

PizzaProgram, 2025-04-19:
```c
#ifdef ARDUINO_PICO_MAJOR
...
#endif
```
**Thanks for the hint!** Yes, this is a bit better. :-)

### I'd just like to mention that since:
- Arduino is building [RP2040 based boards](https://store.arduino.cc/en-hu/products/arduino-nano-rp2040-connect-with-headers),
- also there is "arduino compatible" code for RP2040,
- and nowhere is clearly defined what is and which is AVR / ARCH / MBED / ... and how they differ / correlate ?  

all these compatibility issues generate more and more confusion even for us, mid/pro programmers.
_Now imagine how is it for someone just starting to learn about..._

That's why I've recommended to include "..._LIB".

---

@JAndrassy 
> I would see this platform as the default.

I agree :-)

JAndrassy, 2025-04-19:
there is also `ARDUINO_ARCH_RP2040` which is `ARDUINO_ARCH_` concatenated with the uppercase version of the architecture folder name, which is  .arduino15/packages/rp2040/hardware/rp2040.
I doubt a new alternative RPxxxx platform would choose RP2040 now.

PizzaProgram, 2025-04-20:
> there is also ARDUINO_ARCH_RP2040 

### That naming is also very confusing!
- This library is not runnin only on Arduino IDE. (But on VSC too).
- Does not run on Arduino chips. _(Except maybe [the one](https://store.arduino.cc/en-hu/products/arduino-nano-rp2040-connect-with-headers) I've mentions earlier? Someone please clarify!)_
- Runs not only on chip RP2040 but on **RP2350** too, ... and who knows what else later
- Does not refer to the fact, that this definition is not specifying the:
    - CHIP or 
    - board or 
    - manufacturer, 

but a **multi-purpuse library** to handle more of them!

So the effect of this kind of naming is just the **opposite** what it should represent.

At least the "major" word of `ARDUINO_PICO_MAJOR` refers to something, that suggest that it is not specifying the hardware or manufacturer, and the first 2 result of the search for "arduino + pico" keywords refers to this lib.