earlephilhower/arduino-pico #2920: Unreliable USB/serial connection - add Parity check

PizzaProgram, 2025-04-22:
Please add possiblity for connection to set 0=disabled  1=odd   2=even connection options, not just baud rate.
 (at serialUSB baud is ignored anyway)

## Why?
_In my 30+ years of experience with Serial communication I've never encountered any case, where serial communication was absolutely reliable. POS printers, medical machines, UPSs, etc..._
A small spark, lightning, nearby 240V cables, anything can disturb it, and from there on it:
- either halts completely,
- or starts to send crazy data.
But it helped a lot, if I activated stopbit + EVEN parity by default (+flow control).

Since these boards are connected with USB, flow control is not possible, but I think (and I really really hope,) enabling parity check would help.

Currently I'm experiencing problems with my Pico, running on a 3m cable. I've implemented a LED-blinking pattern, if the serial is not available any more, and usually it happens every each day. _(The record was 2.5 days of working non-stop.)_

PizzaProgram, 2025-04-22:
_Thanks for the quick answer!_
>Can you explain a little more your setup?

I'm currently fixing/upgrading Node-RED's default Arduino/Pico/ESP32 board expander module, based on [ConfigurableFirmata](https://github.com/firmata/ConfigurableFirmata) firmware, which is based on [MIDI protocol](https://en.wikipedia.org/wiki/MIDI). (The process of my improvements are [here...](https://discourse.nodered.org/t/announce-node-red-node-arduino-firmata-remastered-v1-3-4-20250331/96317))

I do not think flow control is possible, if those chips are using serial over USB, because it has only 2 wires to handle RX/TX data.

The main problem is of course that `firmata.js` does not define proper timeout nor would it handle it properly anywhere.
firmata-io.js is using: [node-serial](https://github.com/serialport/node-serialport) to connect to Arduino chips.

earlephilhower, 2025-04-22:
USB CDC isn't a real UART and parity doesn't make any sense for it, nor is it supported in USB.  CDC uses packets with embedded CRC to guarantee data correctness (as do all USB interfaces).  Things like CTS and baud rate are part of the informational flags, but don't really affect host or device.

closed 2025-04-22 by earlephilhower