earlephilhower/arduino-pico #2930: RP2350B - attributing SPI0 RX on GPIO 36 freezes

TadaNoButa, 2025-04-29:
Hi there !

After passing quite a time on this, I've come to the point there must be a problem in the library with the pin 36 for the RP2350b with the SPI (SPI0). It freezes when I do `SPI.setRX(36);` (before `SPI.begin;`).

I've tried with the sdk and `gpio_set_function(36, GPIO_FUNC_SPI);` goes without problem. Thou, I haven't tested further with the sdk to see if the SPI line is really usable.

I have only tested those pins for SPI(0): RX=36, CS=37, SCK=38 and TX=39 and for SPI1: RX=44, CS=45, SCK=46 and TX=43, because it's a custom board, so I can't test other pins (not connected. SPI1 attributions are fine, and for SPI(0): TX, CS and SCK attributions return true, only RX freezes.

Thank you in advance.

closed 2025-04-29 by earlephilhower

TadaNoButa, 2025-04-29:
Thank you as much, Earle !

earlephilhower, 2025-04-30:
No problem!  Bug reports with complete info and simple errors make things easy to fix. :)

PR #2931 (2025-04-29): Update SPI.cppFix RP2350B SPI RX pin list
Fixes #2930
commit 227d71ed (2025-04-29): Fix RP2350B SPI RX pin list (#2931)

Fixes #2930