earlephilhower/arduino-pico #2934: Feature Request: Add support for second QSPI flash on RP2350

HanasakiMana, 2025-05-02:
In datasheet of the RP2350 chip, 2 enternal memory chips can be used by the new QSPI Memory Interface, leading to a maximum of 32MB of emternal memory. Besides the external flash controlled by the CSn pin, you can connect another QSPI chip like NOR flash or PSRAM.

However, arudino-pico only supports functions for PSRAM, which is only designed to interact with a RAM, like malloc(), causing some diffuculties to interact with NOR flash. 

It would be great to add support for second QSPI flash, or even further more, formatting this flash into a dedicated FS to store data.

Greets Mana

HanasakiMana, 2025-05-02:
Thanks for your reply! I just noticed that you added SDIO support after reading your post.

Actually I'm trying to make my own [picoDVI](https://github.com/HanasakiMana/picoConsole) board based on [this fork](https://github.com/adafruit/PicoDVI), just want to design a pixel style game on it. At the first time, I chose SD card to store the assets, but since dvi output has already occupied all PIO state machines, and this project had no support for SDIO back to 2 years ago, so I had to communicate with the SD card using SPI controller, which is really slow. So after reading reading the rp2350 datasheet, the only thought in my mind was using the QMI to store assets into second flash with QSPI connection.

Yes, you are right, a larger RAM is much more important than another flash chip. I can load assets into PSRAM at first and transfer them into the faster puilt-in ram to finish rendering. 

Also, currently I can only choose up to 8MB of the capacity of PSRAM in Arduino IDE, would you mind to add a 16MB option for this? This will really helps. Thank you for your patience!

earlephilhower, 2025-05-02:
Updating `makeboards` to include 16M in the generic-rp2350 case itself is trivial, but it's more complicated than that I think.

Is there a 16MB compatible PSRAM chip?  Does it report the same ID as the common 8MB one?  The [PSRAM code](https://github.com/earlephilhower/arduino-pico/blob/master/cores/rp2040/psram.cpp) needs to be able to identify the chip and setup appropriate timings, and it needs to speak the right commands for the XIP hardware to use it.

HanasakiMana, 2025-05-02:
As far as I know, LY68L6400, a 64Mbit PSRAM chip, has the same commands of Winbond W25Q64, with pin-to-pin swap supported, but I'm not sure about the PSRAM code. Thanks for the information you mentioned, I'll try to draw a pcb to evaluate it.

earlephilhower, 2025-05-02:
There is an Adafruit board with an unpopulated spot for PSRAM.  You might want to grab one of those and the chip and hack a bit before building custom PCBs.

earlephilhower, 2025-05-02:
> Also, APS12804 from APMemory has 128Mbit with QPI supported, but I haven't used it in a real project yet. You can find datasheet [here](https://www.apmemory.com/en/downloadFiles/0324112120wb615978), and I'll also try to evaluate this chip if I can buy some.

A quick comparison to the [datasheet for the 8MB device](https://www.pjrc.com/store/APS6404L_3SQR.pdf) everyone seems to be using shows this has the same command format, which looks good.  The EID[47:45] (section 12.1) will be different (but they don't say exactly *what* it will be on a 128M device) and need to be patched into the [psram.cpp](https://github.com/earlephilhower/arduino-pico/blob/47c2cd2b0be80c357809f7acdfa50cb5b7ccfce4/cores/rp2040/psram.cpp#L177-L189) file here.

HanasakiMana, 2025-05-03:
Oh I see it! Thanks for your patience to check the datasheet and find that!Now I'll try to use this 8.5MB ram space to finish my project, with an SD card attached on it. 

Also, I’m sorry for any confusion or trouble my misunderstandings may have caused, and will make sure to review the documentation more carefully next time.

Best regards,
Mana Hanasaki