earlephilhower/arduino-pico #2935: ADCInput + I2S project loopbak stop working on new plugin versions.

gcallipo, 2025-05-04:
Good morning, I'm using a combination of the ADCInput API and I2S as an output device. In general in the project that uses Raspberry Pico Pi 2, I read a buffer of 128 samples, filter them, and write them back to the I2S output. Everything worked perfectly with previous versions of the plugin (4.3.1) but with the new versions the acquisition of the signal from ADC seems to be extremely slowed down generating continuous noise.

I also tryed to make a simple test of loopbak reading and writing at the same frequencies
using A0 as sampling PIN and PCM5202 as Output DAC.

The samples rate tested was 8000 16000 and more ... 
The BUFFER_SIZE was various 32 ...
The setup:

  i2s.setBCLK(I2S_BCLK);
  i2s.setDATA(I2S_DOUT);
  i2s.setBitsPerSample(16);

  if (!i2s.begin(sampleRate)) {
    while (1); // do nothing
  }

  adcIn.setBuffers(4, BUFFER_SIZE);

  if (!adcIn.begin(sampleRate)) {
    while (1); 
  }

... The loop 

while (adcIn.available() > 0) {

    newSample = adcIn.read();
    
     i2s.write(newSample);
     i2s.write(newSample);
   
  };

All works correctly on 4.3.1 , but not in the new release of plugin and use It with success in several projects in ham radio processing signal.

The complete  project Is : https://github.com/gcallipo/RadioDSP-Pico

Thanks for the great effort to produce this library.
Giuseppe.

earlephilhower, 2025-05-04:
Sorry, I really can't dig into a complete application like that.  Can you make a small, simple and self contained MCVE that shows the problem?  There was a bugfix to the ADC sample rate with multiple channels (it was actually giving data at 2x or 4x the proper speed) but that was 5 months ago or so.  (#2755 )

gcallipo, 2025-05-04:
Ok, I Will create a simple test ADC + direct Output to I2S.  I Will check on 4.3.1 and next in the new version. 
Thanks for replay and waiting for my test .

gcallipo, 2025-05-04:
Here the test, I check some plugin versions on Raspberry Pico Pi 2, latest plugin version that works was the 4.5.1

// Test ADC --> I2S passthrough 
// Test some version plugin:

// 4.5.3 --> FAIL, output very noisy like a square wave
// 4.5.2 --> FAIL, output very noisy like a square wave
// 4.5.1 --> SUCCESS Works, sample and play clean audio
// 4.5.0 --> SUCCESS Works, sample and play clean audio
// 4.3.1 --> SUCCESS Works, sample and play clean audio

#include <I2S.h>
#include <ADCInput.h>

#define BUFFER_SIZE 32

// GPIO I2S pin numbers
// TO connect the PCM5102 DAC
#define I2S_BCLK 15
#define I2S_WS (I2S_BCLK+1)
#define I2S_DOUT 17
const int sampleRate = 8000;

// Create the I2S port using a PIO state machine
I2S i2s(OUTPUT);

// Create inputt on Adc 0 - GPIO26
ADCInput adcIn(26);

void setup() {

  // init adc
  adcIn.setBuffers(4, BUFFER_SIZE);
  adcIn.setFrequency(sampleRate);

  if (!adcIn.begin()) {
    while (1); // do nothing, init fail
  }

  // init i2s
  i2s.setBCLK(I2S_BCLK);
  i2s.setDATA(I2S_DOUT);
  i2s.setBitsPerSample(16);
  i2s.setFrequency(sampleRate); 

  // start I2S at the sample rate with 16-bits per sample
  if (!i2s.begin()) {
    while (1); // do nothing, init fail
  }

int16_t newSample=0;
// execute processing
while(1){
      
    while (adcIn.available() > 0) {
        newSample = adcIn.read();

        i2s.write(newSample);
        i2s.write(newSample);
    };
}

}

void loop() {
   // do nothing here ...
}

earlephilhower, 2025-05-04:
Looking at the differences between 4.5.1 and 4.5.2 shows absolutely no changes to anything remotely connected to this.
https://github.com/earlephilhower/arduino-pico/compare/4.5.1...4.5.2

Could you have some instability in the physical setup that makes tetchy?  Again, there is not a single difference to code on the RP2040 that's being used in I2S or ADCInput.  4.5.2 was mostly the RP2350A/B stuff and minor WiFi related things.

gcallipo, 2025-05-04:
It's strange, I don't think it's directly due to my Ide setup because it was reported to me by various users and I verified it later. I don't have an explanation for this, at the moment we will use the plugin at version 4.5.1, later we plan an I2S input and output solution and this should solve the problem. 
The latest tests that missed Is to check on RP2040 if works, because all my check was on RP2350.

Thanks for the checks. Giuseppe Callipo.

Geekachuqt, 2025-05-05:
Hi, I just want to add that I have encountered the exact same issue with code that uses ADCInput/I2S works fine on 4.5.1 no longer works in 4.5.2. I have also reverted at this point.

earlephilhower, 2025-05-05:
Ah, didn't catch it was the Pico2 and so there was an ADCInput change in 4.5.2.  Basically it was expecting the 2350B GPIOs for ADCInput pins on the 2350A and vice versa because the logic of a `#if` was inverted in the patch.

Please try #2937 and report back.

gcallipo, 2025-05-05:
Great ! I can confirm that the fix works. I updated the 4.5.3 and install manually your fix, and now all works as aspected. Thanks for the excellent support.
I think that the issue can be close.

Thanks again,  Giuseppe.

PR #2937 (2025-05-05): Fix ADCInput masking
Fixes #2935

The PICO_2350A patch flipped the logic of the ADCInput bitmask calcs causing the 2040 and 2350A to use bitmasks for the 2350B (i.e. wrong GPIOs) and vice versa.
closed 2025-05-05 by earlephilhower