earlephilhower/arduino-pico #2936: I2S: Wrong sysClk frequency for sample rate 48 kHz?

ullibak, 2025-05-05:
Hi

A device I use (Waveshare Pico Audio www.waveshare.com/wiki/Pico-Audio) is now produced in a new version with an new DAC using the master clock. So I set the aprropriate parameters in the I2S functions and everything works well for a sample rate of 44.1 kHz. But by no means have I been able to get a sound output using a 48 kHz sample rate.

This is my test code derived from the Master Clock example:

```c
#include <I2S.h>
I2S i2s(OUTPUT);

// GPIO pin numbers
#define pDOUT 22
#define pBCLK 27
#define pWS (pBCLK+1)
#define pMCLK 26  

const int frequency = 440; // frequency of square wave in Hz
const int sampleRate = 48000; 
const int bitsPerSample = 16;
const int amplitude = 1 << (bitsPerSample - 2); // amplitude of square wave = 1/2 of maximum

#define MCLK_MUL  256     // depends on audio hardware. Suits common hardware.

const int halfWavelength = sampleRate / (2 * frequency); // half wavelength of square wave

int16_t sample = amplitude; // current sample value
int count = 0;

void setup() {
  i2s.setSysClk(sampleRate); // best to do this before starting anything clock-dependent

  Serial.begin(115200);
  while (!Serial) {
    delay(10);
  }
  
  i2s.setBCLK(pBCLK);
  i2s.setDATA(pDOUT);
  i2s.setMCLK(pMCLK); // must be run before setFrequency() and i2s.begin()
  i2s.setMCLKmult(MCLK_MUL); // also before i2s.begin()
  i2s.setBitsPerSample(bitsPerSample);
  i2s.swapClocks();
  i2s.begin(sampleRate);

  Serial.printf("sampleRate =  %d, MCLK_MUL = %d, SysClock = %d\n", sampleRate, MCLK_MUL, clock_get_hz(clk_sys));
}

void loop() {
  if (count % halfWavelength == 0) {
    // invert the sample every half wavelength count multiple to generate square wave
    sample = -1 * sample;
  }

  // write the same sample twice, once for left and once for the right channel
  i2s.write(sample);
  i2s.write(sample);

  // increment the counter for the next sample
  count++;
}
``` 

After investigating the code, I found that in I2S.cpp, the system clock is changed according to the sample rate, the function is called "I2S::setSysClk()". In I2S.h, the clock speeds are defined for two classes of sample rates:

```c
    static const int I2SSYSCLK_44_1 = 135600; // 44.1, 88.2 kHz sample rates
    static const int I2SSYSCLK_8 = 153600;  // 8k, 16, 32, 48, 96, 192 kHz
``` 

After some trial and error I found that setting I2SSYSCLK_8 to 150000 (i.e. clock speed of 150 MHz) solved the problem. I was able to get a sound using a sample rate of 48 kHz.

The function I2S::setFrequency() shows how the PIO state machine's clock division factor ist calculated, but I was unable to find any information on how to calculate the correct frequency of the system clock. 

So my question is: Is 153600 for I2SSYSCLK_8 a typo since it is just a permutation of to characters in 135600 given for I2SSYSCLK_44_1?

And what is the correct method to calculate the system clock speed for the sample rates that I2SSYSCLK_8 is used for.

Many thanks!

earlephilhower, 2025-05-05:
Looking at MCLK the multiplier of 256 makes it interesting and possibly your problem.  We need to manually make that MCLK using a PIO state machine.

For 256x MCLK, that's 512 phases at 48k per second = `12,288,000` phases/sec.  `153,600,000 / 12,288,000 = 12.5` which is not possible, you'll get 12 or 13 SYSCLK for the phase time on MCLK.  I'd need to check code, but either one would not match your requested rate.

Try dropping down to MCLKx128 which would give an exact 25.0 SYSCLK/phase.

ullibak, 2025-05-06:
I did not know that "close enough" will do. So I will read the datasheet of the DAC and try to find an appropriate solution.  A first check of the data sheet showed that the DAC does not support a MCLK multiplier of 128x at 48 kHz. 

Thanks for your help!

closed 2025-05-06 by ullibak

ullibak, 2025-05-07:
Just for the sake of completeness (after closing the topic): Here is a short description of how I found an appropriate clock speed for a sample rate of 48 kHz.
Not all clock speeds are allowed. There is a Python script to help finding allowed values mentioned in the SDK documentation, located at src/rp2_common/hardware_clocks/scripts/vcocalc.py

You can run this script with the desired clock speed (in MHz) as a parameter and it returns the nearest possible clock speed or special settings in CMakeLists.txt on how to choose a non-standard clock speed. 

48kHz with a MCLK multiplier of 256x results in a MCLK frequency of 12'288'000 Hz. Aiming at a 12-fold sysClk frequency would yield  147.456 MHz. Now let's use this in the Python script:

```
C:\Clock_Calc>python vcocalc.py 147.456
Requested: 147.456 MHz
Achieved:  147.6 MHz
REFDIV:    1
FBDIV:     123 (VCO = 1476.0 MHz)
PD1:       5
PD2:       2
```
So the nearest possible sysClk is 147.6 MHz. The factor sysClk/MCLK of 12.012 is "close enough" to the desired value and I get sound output. I have been able to obtain a sound with even higher sysClk (150 MHz or factor 12.207), but there is no sound output if the factor is even slightly below the desired value. So obviously (if you can not find an exact solution), you must always use the sysClk frequency that is closest to the desired value on the positive side, i.e. slightly higher.