earlephilhower/arduino-pico #2940: Reassign LED_BUILTIN on the WIZnet W55RP20-EVB-Pico from 25 to 19.

RicardoJCMarques, 2025-05-10:
Currently it's set to the generic Pin 25 that's coincidentally assigned to an actual LED on the pcb. But that LED is itself assigned to be the Ethernet 10/100 Speed LED.

Since there's a User LED on Pin 19 I'd suggest changing to it to make it more user friendly.

earlephilhower, 2025-05-10:
Yes, I see on the schematic that there's a GP19 connection to a separate LED.  I don't think I actually see GP25 listed, but maybe they renamed it internally to the multidie package.  It's a very simple change, why not give it a try?  They actually don't provide any special mapping (see (`variants/wiznet_55rp20_evb_pico/pins_arduino.h`) so you may need to cut-n-paste the contents of `generic/pins_arduino.h` and edit accordingly.

Note they might actually use `PIN_LED` or `LED_BUILTIN` in their driver which this change would break.  I don't have a board but it should be an easy check for you with PCB in hand...

RicardoJCMarques, 2025-05-11:
Hum, I'm sure I can workout how to get to the merge request with the change but how would you suggest I test it first? Lost my mind just trying to figure out how platformio in vs code works with boards like this.

I also tried blinking the other GPIO that are not available as per their list and 25 really is the only one that can blink another LED. It sounds very coincidental and not at the same time.

GPIO17, GPIO20, GPIO21, GPIO22, GPIO23, GPIO24, and GPIO25 they mention as unavailable for internal connections.
16 is connected to the buck-boost converted.
18 is connected to a voltage divider on VBUS.
19 on the LED just noticed also means generic SPI0 as a whole is all assigned wrong too.

In the pinout the peripherals are re-assignable but that's above my pay grade...

earlephilhower, 2025-05-11:
P.IO makes it harder, but since you're not adding a new board it probably isn't too bad.  The issue is where does it keep the installed copy of the core...

There is a way to make it use a `git` clone of the core instead of a release package, but I'm not a P.IO user so can't give any real pointers.

But, if you don't mind hackish things, then just have P.IO dump the full command line for a build.  That should point you to the location where the core is stored, and hacking the specific `pins_arduino.h` can be done there.  Once testted, a real edit/copy over a `git` fork to let you do a PR would be doable.

RicardoJCMarques, 2025-05-11:
Since I'm using VS Code, I could probably just point the Platformio .ini to my development fork's url with the variants/wiznet_55rp20_evb_pico/pins_arduino.h file "fixed" no?
That should get the core to load the assigned pins for this board and see if anything major breaks?

PIO on the w55rp20 is using 4 state machines and should only affect the pins set for it. If I'm not mistaken they are set at libraries/lwIP_w55rp20/src/utility/w55rp20.h

#define WIZNET_PIO_SPI_MOSI_PIN 23
#define WIZNET_PIO_SPI_MISO_PIN 22
#define WIZNET_PIO_SPI_SCK_PIN  21
#define WIZNET_PIO_SPI_CS_PIN   20

What should I do about te SPI0 pins? Just #YOLO some new ones from the pinout that don't overlap with any of the other peripherals on pins_arduino.h?

earlephilhower, 2025-05-12:
> What should I do about te SPI0 pins? Just #YOLO some new ones from the pinout that don't overlap with any of the other peripherals on pins_arduino.h?

Why not use the [default ones in the RPi Pico](https://github.com/earlephilhower/arduino-pico/blob/master/variants/rpipico/pins_arduino.h)?  I don't think they overlap the PIO SPI (but you'd need to check they actually get pinned out...the stacked package might drop them for some reason...)

RicardoJCMarques, 2025-05-12:
The pins assigned for SPI0 [aren't connected to the castellated pins](https://docs.wiznet.io/assets/images/w55rp20-evb-pico-pinout-c7dbe9757fcc123403e7ed81a6984931.png) for general use.
16 is connected to the buck-boost converted.
17 is no where to be found on the schematics so it's internal.
18 is connected to the voltage divider on VBUS.
19 powers the USERLED.

I looked over the pinout and from the generic the only difference is that some GPIO aren't available, importantly those assigned to SPI0.

I just tested loading my fork onto VS Code and setting the LED_BUILTIN to pin 19 didn't break anything during compilation and worked as expected.
Commenting those SPI0 definitions did, should have seen that one coming. I then tried reassigning them to 2,3,4,5 and documented reason behind the change.
They do overlap other peripherals now but that didn't break the core so should be fine? Either that or assigning them to non-existent pins? But that sounds like it could break something unexpectedly.

earlephilhower, 2025-05-12:
That's fine, they can overlap.  Just as long as all the pins are actually legal per the RP2040 spec anything goes.

RicardoJCMarques, 2025-05-13:
Figured, some peripherals would have to be sacrificed.

How should I go about with the  merging now? New branch needed or anything? I've only dealt with merge requests on a tiny kicad library on gitlab...

RicardoJCMarques, 2025-05-13:
[Ok, almost the same.](https://github.com/earlephilhower/arduino-pico/pull/2956)
Since it was a quick change on single file I just edited directly through the webgui. Faster to get going.

I was wondering if I should have documented more on the .h file about other pin assignments like where PIO SPI pins are set in case anyone came looking?
Didn't want to staple on another commit...

earlephilhower, 2025-05-14:
It looks fine to me.  The internal software SPI isn't really something people have control over so I'd say you're all good.  Thanks!

closed 2025-05-14 by earlephilhower